sebanc/brunch #1996: Lenovo IdeaPad Duet 3i cant get Linux to run

mikejkent090, 2024-01-10:
Hello! I recently purchased this tablet off of amazon as a replacement for my Chromebook duet. I needed additional store and a micro SD card reader.

I have been able to get this up and running with brunch , android seems to work fine , but i cant get the Linux environment to set up. IT always just gives me a error when setting up the Linux environment.

I installed using the Ramus image and the latest brunch build.  I have turned on all virtualization in the bios and if i install chrome OS flex Linux sets up fine.

Would like to get this thing up and running with both android and Linux.

The processor is Intel Celeron N4020

Sorry if this is a repeat question or the wrong area , just desperate to get this running before a vacation next week.

Thanks!

tejasraman, 2024-01-10:
What is the error? Can you send me a screenshot and/or a log (obtained using `dmesg`)?
Also, could you try the latest unstable build with the octopus recovery image for troubleshooting purposes?

(P.S about 3 years ago I had an OG duet. Loved the form factor until the screen broke 😄)

mikejkent090, 2024-01-10:
I dont know how to run dmesg? My understanding of brunch is very limited. IVe done basic installs on a few computers not a lot of trouble shooting....

I still have my duet and love it. My only complaint is the storage activating Linux and running android on 32 GB just doesnt cut it

tejasraman, 2024-01-10:
> I dont know how to run dmesg? My understanding of brunch is very limited. IVe done basic installs on a few computers not a lot of trouble shooting....
> 
> I still have my duet and love it. My only complaint is the storage activating Linux and running android on 32 GB just doesnt cut it

I understand that - my OG Duet had only 32 but I used it as an Android tablet and for Word docs. I did anything serious on my Linux desktop.

I asked for 2 things:
- Could you try the latest unstable build with the octopus recovery image for troubleshooting purposes?
- Could you please tell me the error message?

mikejkent090, 2024-01-10:
Sorry about that have flex on there at the moment just downloading the files to attempt a reinstall as we speak. Should I try the octopus image first or reinstall ramus and get the error?

tejasraman, 2024-01-10:
> Sorry about that have flex on there at the moment just downloading the files to attempt a reinstall as we speak. Should I try the octopus image first or reinstall ramus and get the error?

Try octopus and unstable first

guylamar2006, 2024-01-10:
CTRL-ALT-F2 to switch to VT2 since sudo is blocked from crosh, username:chronos,  then sudo dmesg > ~/Downloads/log.txt to view the log in a text editor.

mikejkent090, 2024-01-11:
Thank you for all the help so far. Tried booting octopus with unstable , doesnt boot to the desktop at all or the chrome set up wizard at all just boot loops. 

Im going to re-install ramus and will post the error i get...

mikejkent090, 2024-01-11:
Here is the log from dmesg on rammus
[log.txt](https://github.com/sebanc/brunch/files/13897322/log.txt)

tejasraman, 2024-01-21:
> Here is the log from dmesg on rammus
> [log.txt](https://github.com/sebanc/brunch/files/13897322/log.txt)
> 

Could you try a different kernel? Nothing looks too suspicious here other than some Android binder errors which should not impact Linux (I see those on one of my Brunch systems but Linux works fine)

Sorry for taking so long to respond

closed 2025-12-11 by github-actions[bot]