sebanc/brunch #2001: [Compatibility] Intel Core-i5 3rd Gen (Ivy Bridge) running r119 working well

signigelchan, 2024-01-21:
I tried multiple recovery images on my old laptop (Acer Aspire E1-471G), including `samus`, `rammus`, and `reef`. 
After many tests, `reef-119` finally worked well with `brunch-r119`.

## Hardware Spec
| CPU  | Graphics |
| :---: | :---: |
| Intel Core-i5 3210m  | HD Graphics 4000 and NVIDIA GeForce GT 630M |

---

The following is a brief overview of my relevant test feedback. (only for my current environment): 

| recovery image  | brunch version | Google Play | Linux Env | Crosh Shell | Terminal Mode |
| :--- | :--- | :---: | :---: | :---: | :---: |
| samus-91  | r91  |  :white_check_mark:  | :heavy_minus_sign: | :white_check_mark:   | :white_check_mark:   |
| rammus-119  | r119  | :negative_squared_cross_mark:  | :heavy_minus_sign: | :ballot_box_with_check:  | :white_check_mark:  |
| rammus-120  | r120  | :negative_squared_cross_mark:  | :heavy_minus_sign:  | :ballot_box_with_check:  | :white_check_mark:  |
| reef-119  | r119  | :heavy_check_mark:  | :white_check_mark:  | :ballot_box_with_check:  | :white_check_mark:  |

:white_check_mark: : Working
:negative_squared_cross_mark: : Not working
:heavy_minus_sign: : Not tested
:ballot_box_with_check: : working only partially
:heavy_check_mark: : working by brunch framework option

## Additional
### `rammus-119(120)` with `brunch-r120`
- When starting for the first time after installation, need to configure the brunch framework kernel option **`/kernel-5.15`**, otherwise, it will stuck on the ChromeOS logo .
- `Google Play` not working by brunch framework option **`android_init_fix`** or **`android_init_fix2`** (Something went wrong. Error code: 8) .

### `reef-119` with `brunch-r119`
- `Google Play` working well by brunch framework option **`android_init_fix2`** and **`/kernel-5.15`** .
- `Crosh Shell` using `Ctrl + Alt + T` can open the window, **but pressing any key will automatically close it** .

## Related Issues
- [ChromeOS recovery image to use for your device](https://github.com/sebanc/brunch/issues/307)
- [What's up with the CPU support](https://github.com/sebanc/brunch/issues/1617)
- [Sandy Bridge](https://github.com/sebanc/brunch/discussions/1677)
- [Brunch works on StoneyRidge and 2nd Gen Intel](https://github.com/sebanc/brunch/issues/1797)
- [Sandy/Ivy Bridge CPU support](https://github.com/sebanc/brunch/issues/1894)
- [Crosh shell closing](https://github.com/sebanc/brunch/issues/1691)
- [Crosh shell not opening - illegal Instruction](https://github.com/sebanc/brunch/issues/1362)

signigelchan, 2024-01-21:
**The following is my current brunch framework option configuration**

```
Summary of the configuration:
options: android_init_fix2,mount_internal_drives,no_camera_config,acpi_power_button,suspend_s3
kernel: 5.15
kernel commandline parameters: enforce_hyperthreading=1 
verbose mode: 0
brunch bootsplash: colorful_dark

Brunch configuration saved.
Press any key to reboot your computer...
```

mahaveer3, 2024-01-23:
currently i am using **rammus-120** with **brunch-r120** but why it is showing error while i am trying to dual boot chrome OS 

error: no such device: /chromeos
error: file /chromeos' not found.
error; unrecognized,number-
error: disk loop,7' not found.
error: you need to load the kernel first.

signigelchan, 2024-01-23:
> currently i am using **rammus-120** with **brunch-r120** but why it is showing error while i am trying to dual boot chrome OS
> 
> error: no such device: /chromeos error: file /chromeos' not found. error; unrecognized,number- error: disk loop,7' not found. error: you need to load the kernel first.
> 
> Any Solution for this ?

---

I'm sorry about this, I only used the installation instructions of `Install with Linux` and have no idea about `dual boot` .

lavilao, 2024-01-23:
@signigelchan does still uses arc++ or is it using arcvm?

closed 2024-01-23 by signigelchan

bennchest, 2025-12-06:
> > Actualmente estoy usando **rammus-120** con **brunch-r120** pero ¿por qué muestra un error cuando intento realizar un arranque dual de Chrome OS?
> > error: no existe el dispositivo: /chromeos error: archivo /chromeos' no encontrado. error; no reconocido, número- error: bucle de disco, 7' no encontrado. error: primero debe cargar el kernel.
> > ¿Alguna solución para esto?
> 
> Lo siento por esto, solo usé las instrucciones de instalación de `Install with Linux`y no tengo idea sobre `dual boot`.

puedes compartirme tu iso