sebanc/brunch #2002: file `/chromeos' not found.

mahaveer3, 2024-01-23:
@sebanc  I was trying to dual boot chrome OS but it shows :

`error: no such device: /chromeos.`
`error: file /chromeos' not found.`
`error: unrecognized number.`
`error: disk loop,7' not found.`
`error: you need to load the kernel first.`

plz how to fix this error , i am currently using **brunch r120, rammus_image(chromeos 120)**

My chromeos grub is :

menuentry "Brunch settings" --class "brunch-settings" {
	img_path=/chromeos
	img_uuid=006cf058-b8c4-11ee-8a0b-401c83984f12
	search --no-floppy --set=root --file $img_path
	loopback loop $img_path
	source (loop,12)/efi/boot/settings.cfg
	linux (loop,7)/kernel boot=local noresume noswap loglevel=7 options= chromeos_bootsplash= edit_brunch_config=1 \
		cros_secure cros_debug img_uuid=$img_uuid img_path=$img_path
	initrd (loop,7)/lib/firmware/amd-ucode.img (loop,7)/lib/firmware/intel-ucode.img (loop,7)/initramfs.img
}

lavilao, 2024-01-23:
Should not be a .img after /chromeos?

mahaveer3, 2024-01-23:
I tried .img too but same error

> Should not be a .img after /chromeos?

pwtz-sudo, 2024-01-24:
regarde ceci :Voici une version corrigée de votre configuration Grub :

bash
Copy code
menuentry "Paramètres du brunch" --class "brunch-settings" {
    img_path=/chemin/vers/chromeos
    img_uuid=006cf058-b8c4-11ee-8a0b-401c83984f12
    search --no-floppy --set=root --file $img_path
    boucle de bouclage $img_path
    linux (loop,7)/chemin/vers/kernel boot=local noresume noswap loglevel=7 options=chromeos_bootsplash=edit_brunch_config=1 cros_secure cros_debug img_uuid=$img_uuid img_path=$img_path
    initrd (loop,7)/lib/firmware/amd-ucode.img (loop,7)/lib/firmware/intel-ucode.img (loop,7)/initramfs.img
}
Assurez-vous de remplacer "/chemin/vers/chromeos" et "/chemin/vers/kernel" par les chemins réels de vos fichiers Chrome OS et kernel. Enregistrez les modifications et essayez de redémarrer pour voir si le problème persiste.

mahaveer3, 2024-01-24:
> menuentry "Paramètres du brunch" --class "brunch-settings" {
> img_path=/chemin/vers/chromeos
> img_uuid=006cf058-b8c4-11ee-8a0b-401c83984f12
> search --no-floppy --set=root --file $img_path
> boucle de [...]

where can i find the kernel ?

closed 2025-12-11 by github-actions[bot]