sebanc/brunch #2007: WiFi not found IntelAX101

babolicous, 2024-01-28:
Hi, thanks for the great work. I used the Brunch r120  stable and volteer 120. Installation to usb went fine. On signing in, wifi shows to be inactivated. When i try to activate it still shows no wifi and as soon as i leave the menu it shows to be deactivated again. It is an intelAX101 Wifi Card and a 12th Gen Intel(R) Core(TM) i3-1215U 1.20 GHz. I already tried to enable "iwlwifi_backport" in brunch configuration menu but it is not listed. 
Any ideas? Thanks a lot in advance.

mikefromnj, 2024-06-23:
I am having a very similar issue. Did you ever find a fix?

babolicous, 2024-06-23:
No, so far no solution to the Problem.

OurCodeBase, 2024-06-30:
I have also the similar issues, Anyone solve it please