sebanc/brunch #2011: I can't see the time when you log in to supervised google account

burnsboyamazing, 2024-02-03:
If you can't see the time, there's a screen recording here @windowz414. [Screen recording 2024-02-03 11.35.23 AM.webm](https://github.com/sebanc/brunch/assets/154830140/820c8b11-f019-4163-8f15-a67d15cc923e)
Can't see the time when you open the tray icon, because it's logging in to supervised users.

windowz414, 2024-02-06:
I don't understand why would you ping me for this even though I have never contributed to Brunch outside of the Wiki (hardware compatibility table to be specific), but I'll try to explain this in the most logical way possible.

I have personally never used a supervised account as such accounts are supposed to be given to the children by their parents to prevent them from accessing apps, services and settings that might be dangerous when their immaturity is in consideration - Hence why the kite icon used to symbolize the term in the first place.

The most restricted account type I've used was an organization account provided by my school's IT department, which didn't accept the ToS on Google Admin Console and so provides only the default restrictions provided by Google. I have never faced such a bug even on such an account.

Maybe this is a setting you need to manage in Family Link? Can you log into the supervisor account and check if there are restriction settings for Chromebooks/ChromeOS? At base level, supervised accounts should be prohibited from managing Google accounts linked to the user account, Powerwashing, using Android device link and installing/removing containers (I.e. Google Play Store, which is ARC, and Crostini, the Linux Development Environment) but not have such UI modifications.

I might as well install ChromeOS on a spare laptop and test this out with a throwaway child account I can create using Family Link, but please let me know if you have access to the Supervisor account.

Also a side note: This looks rather like a bug with ChromeOS itself instead of Brunch, so you might as well want to [join the official Discord server made for Brunch users](https://discord.gg/x2EgK2M) and ping me about this in `#chromeos-general` channel after you complete the first experience.

burnsboyamazing, 2024-02-06:
Thanks, I'll check it out.

closed 2024-02-09 by sebanc