sebanc/brunch #2015: grub2 error minicmd.c:138: no such module under Fedora 39

ajbathe, 2024-02-06:
Hello to everybody,,

I successfully installed brunch_r120_stable_20240107 with recovery image chromeos_15662.88.0_zork_recovery_stable-channel_mp-v9.bin under Fedora 39 (Workstation Edition) on a new Framework Laptop 13 AMD 7840U. ChromeOS works really good. I haven't found any problems so far.

But there is a grub2 error message on booting ChromeOS:

> error: ../../grub-core/commands/minicmd.c:138: no such module.
>
> Press any key to continue...

I checked the same scenario under Ubuntu 22.04 LTS and there is not this error. I can't find a solution to supress this (maybe only cosmetic) error.

tejasraman, 2024-02-20:
> Hello to everybody,,
>
> I successfully installed brunch_r120_stable_20240107 with recovery image chromeos_15662.88.0_zork_recovery_stable-channel_mp-v9.bin under Fedora 39 (Workstation Edition) on a [...]

I remember having to manually add a line about TPM if I installed Brunch from Fedora... I'll try this on one of my systems. 

Could you send the grub.cfg that the Brunch grub is using?

ajbathe, 2024-02-20:
```
menuentry 'ChromeOS' --class chromeos {
    rmmod tpm
    img_path=/opt/google/chromeOS/chromeos.img
    img_uuid=8cecbb71-7c5d-4fbe-aba7-b5f2fdb1f6bb
    search --no-floppy --set=root --file $img_path
    loopback loop $img_path
    source (loop,12)/efi/boot/settings.cfg
    if [ -z $verbose ] -o [ $verbose -eq 0 ]; then
        linux (loop,7)/$kernel boot=local noresume noswap loglevel=7 options=$options \
            chromeos_bootsplash=$chromeos_bootsplash $cmdline_params \
            cros_secure cros_debug img_uuid=$img_uuid img_path=$img_path \
            console= vt.global_cursor_default=0 quiet
            #console= vt.global_cursor_default=0 brunch_bootsplash=$brunch_bootsplash quiet
    else
        linux (loop,7)/$kernel boot=local noresume noswap loglevel=7 options=$options \
            chromeos_bootsplash=$chromeos_bootsplash $cmdline_params \
            cros_secure cros_debug img_uuid=$img_uuid img_path=$img_path
    fi
    initrd (loop,7)/lib/firmware/amd-ucode.img (loop,7)/lib/firmware/intel-ucode.img (loop,7)/initramfs.img
}
```

ajbathe, 2024-02-20:
hope this is what you asked for ...