sebanc/brunch #2023: Brunch won't boot on Deck OLED

zackivy, 2024-02-14:
I have a Deck OLED that I got 2 months ago and I wanted to try to experiment installing ChromeOS on a USB/SD card but when I try to boot it up, it gets stuck at the 'Loading Brunch Framework' screen for over 30 minutes and the build I tried was the zork AMD one.

sebanc, 2024-02-16:
Most likely, you need to go in the bios and change "USB Dual Role Device" from "DRD" to "XHCI".

Source: https://www.reddit.com/r/SteamDeck/comments/181kx8b/oled_steam_deck_not_recognizing_usbc_flash_drive/

zackivy, 2024-02-16:
I tried that method, it STILL gets stuck on that screen and IT DOES recognize the drive!

tejasraman, 2024-02-25:
> I tried that method, it STILL gets stuck on that screen for over 30 minutes but IT DOES recognize the drive! (same for SD card which I am trying to install on) ![IMG_0038](https://private-user-images.githubusercontent.com/159971868/305452080-57dc2e1b-420a-4324-a9d4-66db9d9fdbb5.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDg4MzgxNTgsIm5iZiI6MTcwODgzNzg1OCwicGF0aCI6Ii8xNTk5NzE4NjgvMzA1NDUyMDgwLTU3ZGMyZTFiLTQyMGEtNDMyNC1hOWQ0LTY2ZGI5ZDlmZGJiNS5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjI1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIyNVQwNTEwNThaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT01NWE4OTdjZWMyZWZiNGZkN2Y5ZTgxNjNiMzg4NTE2NzI0MGEyMDE4OTFmNDE5NDZjNjE4ZjJlOGFiMGExNWY3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.BjJuo8gV4X6ZD7mG740yn1a6rX8bril4exrLvnC4HOE)

This was patched by @sebanc in the latest unstable

zackivy, 2024-02-29:
So I tried the new build it works, but I cannot get the on screen keyboard to show up when typing and sound is not working.

tejasraman, 2024-02-29:
> So I tried the new build it works, but I cannot get the on screen keyboard to show up when typing and sound is not working.

Onscreen won't show up unless you enable force_tablet_mode in Brunch config (`sudo edit-brunch-config` in a shell or Brunch (settings) in boot menu).

sebanc, 2024-03-01:
Even though it might seem strange, the on screen keyboard not working is a consequence of sound not working.

I have tried to add the necessary firmwares/configs for sound to work on the oled steam deck but the alsa version used in ChromeOS might just be too old.

I can try to investigate further if someone provides the output of both "dmesg" and "cras" commands in crosh shell

closed 2025-12-11 by github-actions[bot]