sebanc/brunch #2031: Multitouch Surface 5 Pro not working

Djrush86, 2024-03-01:
After install the actually Brunch and Chrome OS I can only touch Wie one finger, no multi touch. Is there any solution for that?

tejasraman, 2024-03-04:
> After install the actually Brunch and Chrome OS I can only touch with one finger, no multi touch. Is there any solution for that?

Do you have ipts_touchscreen enabled in framework options? That's required for full IPTS multitouch on some Surface Pro devices (Surface Pro 4/5/6)

Djrush86, 2024-03-05:
Yes I have. I am currently on a older brunch r113 and rammus build r13. All is working fine here. No problem.

I will install this week a newer build and send the dmesg file from it.

tejasraman, 2024-03-05:
> Yes I have. I am currently on a older brunch r113 and rammus build r13. All is working fine here. No problem.
> 
> I will install this week a newer build and send the dmesg file from it. 

Thanks so much @Djrush86

closed 2025-12-09 by github-actions[bot]