sebanc/brunch #2032: no multi touch and surface pen support on surface pro6

Ken880408, 2024-03-02:
Already enable ipts, but surface pen and multi touch function still not work on my surface pro 6
chrome os build is 120

Ken880408, 2024-03-02:
Try all kernel and update to V121, still only one point touch, sudo /usr/bin/iptsd/ shows Devices is Requerd.

tejasraman, 2024-03-04:
See #2031 - this is a widespread issue for sure. 

If you could send me a dmesg I'd be happy to investigate this issue further. To obtain a dmesg log, type `sudo dmesg > ~/Downloads/dmesg.txt` into a shell and put the dmesg.txt in your Downloads on this issue.

Ken880408, 2024-03-05:
[dmesg.txt](https://github.com/sebanc/brunch/files/14494644/dmesg.txt)
here is the file, hope you can find some solution for this issue....

Djrush86, 2024-03-06:
Same Problem here on my Surface 5 Pro #2031

Ken880408, 2024-04-02:
with unstable brunch R122, the problem is fixed ! thanks for your hard work !

Ken880408, 2024-04-04:
After more test, I found that if I put my device to sleep for maybe 10 minutes, the multi touch and pen function will lost again, need to restart system to bring back multi touch, so I let this issue open again.

closed 2026-01-07 by github-actions[bot]