sebanc/brunch #2040: Boot problems after an update to ChromeOS v122

MarioNooo, 2024-03-17:
Before updating to 122.0.6261.118 Brunch worked perfectly no other issues, but after the update to that version it will just restart after the Chrome OS splash logo.
Here is my PC's specs:

- CPU: AMD Ryzen 5 5600G
- GPU: Cezanne [Radeon Vega Series / Radeon Vega Mobile Series] (integrated)
- Motherboard: Asus PRIME B550M-A (WI-FI)
- RAM: 16 GB DDR4
- Storage: HP SSD EX900 500GB 

What I tried:
1. Enabling verbose boot to find a kernel panic (Couldn't find anything)
2. Updating brunch-unstable every day
3. Using markarth instead of gumboz recovery image 

Anything what I did above didn't work.

dazbeattie, 2024-04-15:
I had something similar,  change the kernel to 5.15, enable pwa and updates in the patches and see how it goes. Once I did this, the machine booted, then was able use kernel 6.6 and the normal patches I use.

MarioNooo, 2024-04-15:
Nope still won't work, even after updating brunch. But now  I noticed on  debug boot  this message:
losetup: /fake_loop: Warning: file is smaller than 512 bytes; the loop device may be useless or invisible for system tools.
Can it be related to by problem? 🤔

closed 2025-12-09 by github-actions[bot]