sebanc/brunch #2047: how to install chromebrew on branch framework? how to enable developer mode?

thekarananand, 2024-03-24:
https://github.com/chromebrew/chromebrew

 - how to install chromebrew on branch framework?
 - how to enable developer mode on brunch framework?
 - how to get access to sudo and other utilities working?
 - how to access terminal / crosh ?

onepointfive-REAL, 2024-05-04:
crosh: chrome-untrusted://crosh

lucasgabmoreno, 2025-05-01:
Get intro VT-2:
`Ctrl+Alt+F2`

Login:
Localhost login: `chronos`

Change sudo password:
`sudo chromeos-setdevpasswd`

Dev mode:
`sudo crossystem dev_boot_signed_only=1`
(Doesn't work, returns errors)

Install chromebrew:
`bash <(curl -L [git.io/vddgY](http://git.io/vddgY)) && . ~/.bashrc`
(Stucked here! Returns error)

Back to ChromeOS:
`Ctrl+Alt+F1`

---

Into ChromeOS Flex all works into chronos shell after "sudo crossystem dev_boot_signed_only=1" setted in VT-2

closed 2025-12-07 by github-actions[bot]