sebanc/brunch #2053: Stuck on Applying Patches since upgrade to Brunch 122

dazbeattie, 2024-04-09:
Using a Surface Pro 6 as my "Chromebook". Was on Brunch 121 stable and ChromeOS 121. I have always updated Brunch then ChromeOS. Just ran the PWA to update Brunch to 122 stable. I am now stuck on "Applying Patches" screen and will not proceed any further. I have reverted to kernel 6.1 too and the same error occurs. Is there an easy way to rollback to Brunch 121 to get my machine back? It is a dual boot, so will not take me too long to rebuild if required.

jaimelalone, 2024-04-10:
I also experienced this, but I had verbose boot enabled.  I kept getting "decompression failed with status 6".  What eventually fixed it for me was going in and modifying the boot settings/patches.

dazbeattie, 2024-04-10:
> I also experienced this, but I had verbose boot enabled. I kept getting "decompression failed with status 6". What eventually fixed it for me was going in and modifying the boot settings/patches.

What did you modify in the boot settings / patches ?

jaimelalone, 2024-04-10:
> > I also experienced this, but I had verbose boot enabled. I kept getting "decompression failed with status 6". What eventually fixed it for me was going in and modifying the boot settings/patches.
> 
> What did you modify in the boot settings / patches ?

I don't remember what I had set previously, but I tried rolling back the kernel to 6.1 as well, but it didn't work.  I think I left it bare stock patches.  I'll troubleshoot later today to see if there is a particular patch I forgot to enable that is causing an issue.

sebanc, 2024-04-10:
Did you only update brunch or have you updated ChromeOS also ?

jaimelalone, 2024-04-10:
> Did you only update brunch or have you updated ChromeOS also ?

When I updated yesterday via PWA, only brunch r122 stable was updated.  I am still running recovery image rammus R121

dazbeattie, 2024-04-10:
> > Did you only update brunch or have you updated ChromeOS also ?
> 
> When I updated yesterday via PWA, only brunch r122 stable was updated. I am still running recovery image rammus R121

This is what I had done too. I always update Brunch stable before updating ChromeOS.

dazbeattie, 2024-04-10:
> I don't remember what I had set previously, but I tried rolling back the kernel to 6.1 as well, but it didn't work. I think I left it bare stock patches. I'll troubleshoot later today to see if [...]

Appreciate it. I normally ensure the following patches are enabled.

Enabled_Updates
PWA
IPTS_touchscreen

dazbeattie, 2024-04-10:
> Right now I am not really sure what could cause this, if you have an usb flashdrive available I would be interested to know if you have the same issue with a fresh install of brunch stable r122.

I will run the fresh install when back at home, using the normal process for dual booting with Windows. Will respond when I have this done. Will be a day or so as I am away.

Thanks again for your work and support

clee604, 2024-04-11:
I had same issue where it was stuck on applying patches after a brunch only upgrade using PWA.

It wouldn't boot after that until I went back into the brunch configuration, selected the same options that I think I had before (default kernel), PWA updates and nothing else. It rebuilt and booted up after that fine.

dazbeattie, 2024-04-12:
Thanks @clee604 

I have enabled pwa and updates only, used kernel 5.15 and it booted.

Edited the config back to normal 6.6, pwa, updates and ipts. Booted successfully again.

Never had to downgrade the kernel and then backup again. But you learn something new everyday. 

I have kept Brunch at 122 and for testing upgraded ChromeOS to 123 and the system is working fine again.

dazbeattie, 2024-04-12:
Downgraded to kernel 5.15 and only enabled pwa and updates, system booted fine. Then was able to upgrade kernel to 6.6 and enable other options fine.

dazbeattie, 2024-04-12:
Downgraded to kernel 5.15 and enabled pwa and updates. System booted fine. Updated to kernel 6.6 and enabled ipts. System booted up fine again.

closed 2024-04-12 by dazbeattie