sebanc/brunch #2058: Can you build against outdated recovery with new brunch script?

madhits, 2024-04-15:
I found a system that closely matches my netbook. See: Acer Chromebook 14 (CB3-431) https://cros.tech/device/edgar/

Mine is a Atom N455 GMA 3150. Currently running cloudready 96. I'd like to get a few numbers ahead of 96 if I can for my offline system. I am trying to run VLC to watch local network DLNA stream with this system. Chrome OS 103 would be perfect. 

So My question is 2 fold. 

1. Can I build with current brunch 122 against 103 recovery? Or should I use brunch 103?

2.Also do the newer brunch versions work with the older MBR_support brunch.tar ?

My guess is the answer is I need to use 103 because that still supports old MBR bios boot up. I know that my machine should be e-waste, but with our planet turning into a plastic e-waste mess (65 tons of it in the ocean) Im trying to keep this system which still gets 6hrs battery life still going. 

I know MBR/Bios boot is no longer supported and I know I'm going to be told to go to linux where support lives on, but chrome os cloudready is amazing on this system, so I hope I can find a recovery that might work for me and push me a few numbers ahead.

THANKS

Tajudeen0, 2024-12-16:
Please have you tried it yet?

ksandbergfl, 2025-01-08:
> Please have you tried it yet?

I tried it this week.  I have a CB3-532 (BANON) with a Celeron N3060, it expired with ChromeOS v103.    I downloaded Brunch 103, and built a USB Brunch boot using a v118 image from CORAL.   It works!   I have audio and everything.   I can't get it to work beyond v118 tho...   v120, v126, v130 don't work.   It appear to be some video driver issue

ksandbergfl, 2025-01-13:
> Update: Edgar with 122 and MBR almost boots but wont launch the chrome OS installer.. Going to try "coral" as it seems >like it might be the latest which still has support for atom cores with i915 GPU driver. WIll update, still looking for any >feedback or direction.

I just completed installing Brunch v126 with ChromeOS v126 CORAL image onto my old Acer CB3-532 (Celeron n3060).   Everything is working fine, except for webcam.  I'm very pleased, at the moment.

closed 2025-12-07 by github-actions[bot]