sebanc/brunch #2066: Brunch on Intel Core 7 Ultra laptop

krishnarao-1, 2024-05-10:
Hi,
I have lenovo thinkbook with Intel Core 7 Ultra and I tried to install chrome os as image for dual boot first using the voxel image and later using a screebo image.
No luck and all I get is a black screen with no messages/nothing. 
I have waited nearly an hour and saw no progress.
Many thanks,
Krishna

SLEDTU, 2024-07-28:
Same here; MSI Prestige Evo 13 A1 - U7-155h, tried Brunch & Flex.

cemalgulluoglu, 2024-10-02:
I am having the same issue. I have a dualboot installation. Neither Brunch nor Brunch Settings works. I have a HUAWEI MateBook 14 (2024) with Intel Core Ultra 7 155H processor, 16 GB memory and 1 TB disk. Tried with volteer, rammus, hatch, brya and rex images but no luck. In case I missed something with the configurations or Grub, I installed Linux Mint to another partition and am able to boot properly via the same Grub instance.

Addiitonal information:

- Secure boot disabled
- Fast startup is disabled
- Hibernation is disabled
- The brunch partition is unencrypted, no Bitlocker
- Tried with a) 200 GB partition and 198 GB image size, and b) 66 GB partition size and 60 GB image size
- Tried all botth with and without the -l (legacy EFI bootloader ) parameter as suggested in the troubleshooting-and-faqs.md

Schefflin, 2024-10-27:
What kind the REimg should be chosen for ultra series1?

krishnarao-1, 2024-10-29:
So far, nothing has worked for me
Tried volteer, rammus, rex, brya, pujjo

numbqq, 2025-01-09:
I have the same issues on Intel Core Ultra cpu, how to fix this?

cemalgulluoglu, 2025-02-02:
> [@cemalgulluoglu](https://github.com/cemalgulluoglu) Did you have done it?

I have been trying with every stable release and as of version version r131 stable 20241226, it still doesn't work. Referencing this site: https://www.chromium.org/chromium-os/developer-library/reference/development/developer-information-for-chrome-os-devices/ I'm trying with Rex image as Ultra Core 155h platform is Meteor Lake. I'm open to test any image if it helps.

Snape3058, 2025-04-28:
@cemalgulluoglu I think the rex image should work, as it is the image of ASUS ExpertBook CX54 Chromebook Plus Enterprise (CX5403), which has an Intel Core Ultra 7 155h CPU originally. However, the latest release of brunch does not support writing it to a USB drive as the image contains 14 partitions rather than 12. Do you know how to do it manually? thx

stevierayg, 2025-05-21:
@Snape3058 I don't see 14 partitions in the rex recovery image, please explain.  The shell script that loops to write the modified brunch image is in chromeos-install.sh, which you could modify on your machine before running it.

Here is where I see 12 partitions in the rex recovery image: 
$ fdisk -l chromeos_16181.61.0_rex_recovery_stable-channel_RexMPKeys-v5.bin
Disk chromeos_16181.61.0_rex_recovery_stable-channel_RexMPKeys-v5.bin: 4.3 GiB, 4621046272 bytes, 9025481 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: E925551B-376C-054B-B3C7-37D7CE1A0482

Device                                                               Start     End Sectors   Size Type
chromeos_16181.61.0_rex_recovery_stable-channel_RexMPKeys-v5.bin1  8495104 8763288  268185 130.9M Linux filesystem
chromeos_16181.61.0_rex_recovery_stable-channel_RexMPKeys-v5.bin2   405504 1454079 1048576   512M ChromeOS kernel
chromeos_16181.61.0_rex_recovery_stable-channel_RexMPKeys-v5.bin3  1454080 8417279 6963200   3.3G ChromeOS root fs
chromeos_16181.61.0_rex_recovery_stable-channel_RexMPKeys-v5.bin4  8417280 8482815   65536    32M ChromeOS kernel
chromeos_16181.61.0_rex_recovery_stable-channel_RexMPKeys-v5.bin5  8482816 8486911    4096     2M ChromeOS root fs
chromeos_16181.61.0_rex_recovery_stable-channel_RexMPKeys-v5.bin6   262208  262208       1   512B ChromeOS kernel
chromeos_16181.61.0_rex_recovery_stable-channel_RexMPKeys-v5.bin7   262209  262209       1   512B ChromeOS root fs
chromeos_16181.61.0_rex_recovery_stable-channel_RexMPKeys-v5.bin8   266240  274431    8192     4M Linux filesystem
chromeos_16181.61.0_rex_recovery_stable-channel_RexMPKeys-v5.bin9       64  262207  262144   128M unknown
chromeos_16181.61.0_rex_recovery_stable-channel_RexMPKeys-v5.bin10 8763289 9025432  262144   128M unknown
chromeos_16181.61.0_rex_recovery_stable-channel_RexMPKeys-v5.bin11 8486912 8495103    8192     4M Linux filesystem
chromeos_16181.61.0_rex_recovery_stable-channel_RexMPKeys-v5.bin12  274432  405503  131072    64M EFI System

Partition table entries are not in disk order.

Snape3058, 2025-05-22:
@stevierayg The recovery image I downloaded is newer than yours, 16209.50.0, whose output of `fdisk -l` is presented below. I found this problem because the brunch script (stable_20250421) complains that "input.bin is not a valid ChromeOS recovey image", which fails the condition check `[ $(cgpt show -i 13 -b "$source") -gt 0 ]`. With a further check on my recovery image, I found that the size of the 13th partition is not zero because it contains 14 partitions.

In addition, I also tried to use [balena-etcher](https://github.com/balena-io/etcher) to write the recovery image directly to my u-disk, but it also fails to boot.

```
Disk chromeos_16209.50.0_rex_recovery_stable-channel_RexMPKeys-v5.bin: 4.3 GiB, 4621054464 bytes, 9025497 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: 231D4BFE-E1AB-D546-9619-BBAD53C7FF42

Device                                                               Start     End Sectors  Size Type
chromeos_16209.50.0_rex_recovery_stable-channel_RexMPKeys-v5.bin1  8495104 8763304  268201  131M Linux filesystem
chromeos_16209.50.0_rex_recovery_stable-channel_RexMPKeys-v5.bin2   405504 1454079 1048576  512M ChromeOS kernel
chromeos_16209.50.0_rex_recovery_stable-channel_RexMPKeys-v5.bin3  1454080 8417279 6963200  3.3G ChromeOS root fs
chromeos_16209.50.0_rex_recovery_stable-channel_RexMPKeys-v5.bin4  8417280 8482815   65536   32M ChromeOS kernel
chromeos_16209.50.0_rex_recovery_stable-channel_RexMPKeys-v5.bin5  8482816 8486911    4096    2M ChromeOS root fs
chromeos_16209.50.0_rex_recovery_stable-channel_RexMPKeys-v5.bin6   262208  262208       1  512B ChromeOS kernel
chromeos_16209.50.0_rex_recovery_stable-channel_RexMPKeys-v5.bin7   262209  262209       1  512B ChromeOS root fs
chromeos_16209.50.0_rex_recovery_stable-channel_RexMPKeys-v5.bin8   266240  274431    8192    4M Linux filesystem
chromeos_16209.50.0_rex_recovery_stable-channel_RexMPKeys-v5.bin9       64  262207  262144  128M unknown
chromeos_16209.50.0_rex_recovery_stable-channel_RexMPKeys-v5.bin10 8763305 9025448  262144  128M unknown
chromeos_16209.50.0_rex_recovery_stable-channel_RexMPKeys-v5.bin11 8486912 8495103    8192    4M Linux filesystem
chromeos_16209.50.0_rex_recovery_stable-channel_RexMPKeys-v5.bin12  274432  405503  131072   64M EFI System
chromeos_16209.50.0_rex_recovery_stable-channel_RexMPKeys-v5.bin13  262210  262210       1  512B unknown
chromeos_16209.50.0_rex_recovery_stable-channel_RexMPKeys-v5.bin14  262211  262211       1  512B unknown

Partition table entries are not in disk order.
```

rickythebeast, 2025-06-18:
any luck on this issue?

cc-kyoto, 2025-07-30:
Also blank screen here after selecting entry in Grub here on Meteor Lake Intel Core Ultra 5 134U (Dell Latitude 7350 Detachable)
Have tried the latest recovery (137) of voxel and rex with current (Brunch r137 stable 20250616)

Noted the similar issues raised for Meteor Lake on ChromeOS Flex and Fyde. 

Guessing it is due to the updated onboard ARC graphics? Linux (Ubuntu) runs great though - rex seems to be the only official one for Meteor Lake based chromebooks available at the moment, maybe we get more luck when newer Lunar Lake ones become available

SLEDTU, 2025-09-14:
Just booted in, and installed ChromeOS flex fine - runs ok, wakes from suspend. 

No touchpad and no wifi after suspend on my MSI Prestige 13 AI (155h).

SLEDTU, 2025-10-06:
[image]

Sorry for the delay - here is a screenshot of version and specs. 

I just used the Chromebook Recovery Utility and selected Google/ChromeOS Flex and booted in. Everything seems to work except audio for me, but BT audio works. Trackpad requires Dev channel. 

Here's my laptop for reference; 
|https://au.msi.com/Business-Productivity/Prestige-13-AI-Evo-A1MX

TestMyLox, 2025-10-24:
Dos anyone have an idea how to install ChormeOS+Bruch or ChromeOSFlex on a DELL Pro 16 Plus (PB16250) with a DELL Core ultra 235U?
I turned off secure boot already but I´m stuck on the Brunch Start Screen where you can select "ChromeOS / ChromeOS (settings).

cc-kyoto, 2026-09-14:
Just a note I managed to get mine to boot brunch recently via the modifications at:
[brunch-lunar-lake](https://github.com/kk1987/brunch-lunar-lake)
Successfully boots via kernel 7.1 using volteer/voxel recovery 152