sebanc/brunch #2072: Latest updates:  cannot connect to Android hotspots (Surface Book 2)

1kenthomas, 2024-05-24:
Howdy,  running Brunch on Surface Book 2,  latest rammus.

For the past 2 updates,  I have been unable to connect to a WiFi hotstop (or shared data) on Android hotstops.   (Have tried many phones and cycled through various kernels & drivers,  no luck).

Other hotspots work fine.

If anyone has any idea of the direction to point me into,  it would be much appreciated.  Thanks.

millerjryan, 2024-05-25:
I cannot even connect on the latest update with a Surface book 2.   It keeps throwing a bad password.

1kenthomas, 2024-05-26:
> I cannot even connect on the latest update with a Surface book 2. It keeps throwing a bad password.

It sounds like we have the same problem,  n=2 now,  thanks for the reply/confirmation.

FWIW I've tried kernels 5.15 with the top half of the WiFi drivers,   no go (and a bunch of the other kernels).   USB tether is available from the non-USB-C ports if you don't know,  which is an OK work-around.

I'll continue to go through kernel and driver combos if I have time,  but suspect that will not solve.

Any other hardware experiencing a similar problem?

millerjryan, 2024-05-27:
So through my testing, I have done the follow.

Brunch r121 with ChromeOS 121            .... works
Latest Brunch r123 with ChromeOS 124 .... failed to connect
Latest Brunch r123 with ChromeOS 123 .... failed to connect
Latest Brunch r123 with ChromeOS 121 .... works

Ken880408, 2024-05-29:
Same issue here, with surface pro 6, Any wifi with password will return bad password and not able to connect, but I can connect to my android hotspot if I set it no password, maybe you can try it for a short solution.

Brunch r124 0511 with chrome OS 124