sebanc/brunch #2078: Compressed iwlwifi firmware fails to load on chromebook-4.19 kernel.

MarvinOl, 2024-05-30:
I am using ASUS Chromebook Flip C302 with brunch. As the device is EoL for official support this gives it extended life. Because of the Chromebook audio implementation I am forced to use 4.19 kernel - the only one where Chromebook built-in audio works.

The Asus C302 has Intel 7265D wireless card for what iwlwifi tries to load /lib/firmware/iwlwifi-7265D-29.ucode. But that does not exist, it only exist as compressed /lib/firmware/iwlwifi-7265D-29.ucode.xz. The fix is a simple "unxz /lib/firmware/iwlwifi-7265D-29.ucode.xz" and on next reboot the wireless works. But the fix must be applied after every update, when rootfs is being rebuilt the uncompressed file is removed.

As wifi does work out of the box on newer kernels (but the built-in audio does not) I think the problem is a more general one - iwlwifi on 4.19 kernel does not work with compressed firmware files.

This is not a major showstopper but inconvenience that needs manual fixing on the device after every update.

MarvinOl, 2024-06-16:
Upgraded to latest brunch release and all the iwlwifi firmware are uncompressed. So problem no longer exist.

closed 2024-06-16 by MarvinOl

huLter, 2024-07-11:
@MarvinOl  Sorry to bother you. I'm also planning to install brunch on my c302. which chrome os recovery are you using please?Is there any necessary settings in brunch?