sebanc/brunch #2079: Using a Chromebook Plus image

MrDanMak, 2024-05-31:
I'm looking to see if using a Chromebook Plus image will allow me to use Chromebook Plus features? I have a Surface Pro 9 and I'm using a Brya image which is working fine, but even on the newest 125 software I only get the regular ChromeOS features and not the Plus features. The only thing not working is the camera which is a Linux Surface kernel issue anyway. 

It only works when using the 6.1 Kernel, so I wonder if this is why I'm not getting the Chromebook Plus features? Otherwise the only thing I can think of is that the Brya image I used is for a device that doesn't support Chromebook Plus (I'm sure I confirmed beforehand however).

I guess my main question is if it's possible to use my device as a Chromebook Plus?

kurtbahartr, 2024-06-03:
TL;DR Have you tried to check if you have the Plus features that don't require the presence of a cam (I.e. magic eraser for now)? I'll also look into this two weeks later when I get more spare time in my hands.

-----

I'm honestly surprised to randomly come across someone with the exact same idea as I had in the day you've posted this, which is you. I have a Casper Excalibur G770.1245 which packs an Intel Core i5-12450 which in theory should be able to boot the `brya` image. However, it was only stuck on ChromeOS logo forever because it also packs an RTX 3050 Mobile GPU, which for some goddamn reason also triggers the kernel to modprobe the `nouveau` driver and make the entire UI renderer just crash. It's all fine with `volteer` board's images however and I'm currently using `voxel` image daily as a single boot, it doesn't seem to have a trace of that driver anywhere at all...although the base devices of that board are not Chromebook Plus laptops so that's be a huge tradeoff.

My finals started today and I managed to have some spare time to search online to see if anyone thought of, let alone attempted to, get such an image booting, let alone running just fine. I'll most probably end up diving into the Arch Wiki and see if there's a kernel parameter that would prevent it from loading a module, which is `nouveau` in my case, and have that image boot up on my laptop.

However, considering *your* case, Google might have some stricter checks against hardware for providing the Plus side. You've mentioned the camera doesn't work due to the kernel and the lack of it is just enough for Google to block it all out considering Gemini, which is also a part of Plus and potentially replaces Assistant, also has an option to take photos of things with your camera at least on the web version. Can you see if you have any other Plus features such as the magic eraser?

I'll start doing more intensive personal tests by the end of my finals, which is two weeks later. Until then, I can try to help you troubleshoot it on your end blindly and see where we can go that way.

MrDanMak, 2024-06-05:
I can use some features due to my Google 1 subscription, but those are only features for that product.

I get no Gemini AI features or the other features mentioned being compatible with a Plus device on the 125 update. Like I mention, if I boot a brya image using the 6.1 kernel it works fine, but not with 6.6. I do notice however, that audio does not work, so clearly there are other hardware issues with the brya image and my device. Whether that's with the Linux Surface Kernel or with the brya image I'm not exactly sure.

For now I've reverted back to a voxel image with every standard ChromeOS function working fine (minus the camera as that's a Surface issue).

kurtbahartr, 2024-06-05:
I've done some further investigation into this last night and I tried every single kind of combination of patches that I could think of. However I've noticed one single issue and that is the Brunch Framework cannot do its patches properly at least with the `brya` image that I have used (I used `kano` if it's of concern): `Could not write file: No space left on device`. This usually happens when there's no longer space in the target partition and my best bet in this case is Google is trying to go the same approach as they are going with Android - You know virtual partitions have been introduced and whenever you tend to flash a GSI in a virtual partition, fastbootd first resizes that partition to be the size of the GSI image and then flashes the image.

@sebanc, I think it's about time we go for some kind of re-partitioning in post install (or during RootFS rebuild within Brunch Framework) so there's enough space in the target root partition for the patches to be applied correctly just like on Android, in which the developers set the `BOARD_<part>_PARTITION_SIZE` (or something like that, I kinda forgot) to a higher value than what the partition would normally be in `BoardConfig.mk` while staying within the total size of the `super` partition so users could flash additional packages like GApps in case of a vanilla ROM or in case users would want to flash something like Apatch to gain root access. What do you think?

closed 2025-12-05 by github-actions[bot]