sebanc/brunch #2092: having issue with play store  in my chrome os

FZY1822, 2024-06-27:
I have Lenovo V14 with amd athlon silver 3050u
running os zork 125 stable berknip and brunch 125r
please help me to solve  issue with  my playservices and playstore
I'm trying 
[image]

dadyal, 2024-07-09:
i am also having issues with play store on ThinkPad T480  and I have intel CPU  I get error code 7 never opens or updates play store

onyxsolo, 2025-01-25:
Did you ever find a solution?  I'm having the same issue using several versions up to and including brunch/zork r131.  I have a ryzen gen 4 processor and tried all the available kernels.  Kernel 6.1 seems to work the best when it comes to functionality but still give me an error code 5 when trying to launch the play store.

closed 2025-12-04 by github-actions[bot]