sebanc/brunch #2106: Stuck at chrome os logo on first boot

Mir5382, 2024-07-14:
Installed chrome os from rammus board and then leona version 126 on dell Inspiron 3542 i5-4210u....
Reinstalled once and rebooted few times but still no luck 
[image]

Yassingames, 2024-07-17:
I have the same laptop and same issiues but i am an i74500u and my nvidia gpu does not work properly please help i installed chromeos r101 a year ago i was so happy with it but i was using windows at that time and i was dualbooting it was in a ntfs partition and one day   i neede to print some school work i pluged my flash drive but it wasnot formatted properly because i had a previous installtion on it but i never noticed that so i went to my drive where chromeos is installed and thought it was the flash drive i hit shift delete in the chromeos inage ab=nd was never able to restore with software i was plannig to try linux so after many failed attempts to restore the file i quit and tried kali linux for 11 months and then last month i was tired from kali and wanted to switch to another disto i tried to install blendos but it would not install then i tried to install deppin which i am using for a month i tried install chromeos r126 again but it stuckon chromeos logo for a long time i tried r125 too and this week i tried another attempt for r126 i left it for 72 hours an hour ago irestarted to send an issuie and i found yours try installing an older version like r101 then updating i will try it but i have internet issiues so my download speeds are slow thanks and i need help too

Yassingames, 2024-07-17:
Hello back there i just downloaded brunch r 101 and chromeos r101 from chrome100 and tried installing it and boom 2 minutes and it worked i think i am gonna update now i am gonna learn how to do that i am now on chromeos and everything works perfectly fine

Mir5382, 2024-07-22:
> r101

Did u update then???

Yassingames, 2024-07-22:
yep but did not work i tried r109 and r120 is it your first time did you download other versions are you using r101 now???

Mir5382, 2024-07-22:
Oh sad..... I installed the ones you recommended r101 and chromeos 101  but downloaded r126 and chromeos 126  yet to be installed......coz of confusion in the method i saw on YouTube

Mir5382, 2024-07-22:
> yep but did not work i tried r109 and r120 is it your first time did you download other versions are you using r101 now???

Bro r126 failed.....let me know if any latest one works for u...

Yassingames, 2024-07-22:
ok

Yassingames, 2024-07-22:
i am trying r 108 now the first version of chromeos in 2023 so far nothin other than r101 works i tried r126 and r125 and r109 and r120  i am now trying r108 if you have discord give to help you there 
[image]

Yassingames, 2024-07-22:
[image]
i gonna try r 108 now if it worked i will comeback

Mir5382, 2024-07-22:
> ![image](https://private-user-images.githubusercontent.com/91526655/351080042-a911bd25-e040-4916-b4de-019858171052.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

Ok

Yassingames, 2024-07-23:
r108 doesnot work neither r102  trying r 113

Yassingames, 2024-07-23:
tried r 113 till now it works without playstore and linux is having trouble restarted twice for no reasan lag 
[image]
[image]
[image]
it crashed github saying no ssh certificate

Yassingames, 2024-07-23:
[image]
****

Mir5382, 2024-07-23:
> tried r 113 till now it works without playstore and linux is having trouble restarted twice for no reasan lag [...]

It is strange btw...... But this gpu thing is messing up everything 😔...
Actually I want to upgrade from 101 coz few apps crash in playback..

Yassingames, 2024-07-23:
Well what gpu thing u mean the dgpu mine is broken maybe that is the problem i will check a fix but r113 does not run google playstore or linux sadly

Yassingames, 2024-07-23:
**Please send me your full device data + contact me on discord my account (anonyoumsdevhacker)**

Mir5382, 2024-07-23:
> **Please send me your full device data + contact me on discord my account (anonyoumsdevhacker)**

I don't use discord....
Plus mine is i5 4210 dell Inspiron 3542 Nvidia GeForce 820M 
4GB 1TB HDD

Yassingames, 2024-07-23:
i have same gpu 840m he is the problem i found a method i will try it

Mir5382, 2024-07-23:
> i have same gpu 840m he is the problem i found a method i will try it

Ok u try and let me know too

Yassingames, 2024-07-23:
i think i found trying r 126 with the fix wait for results i will tell you the method

Mir5382, 2024-07-23:
> i think i found trying r 126 with the fix wait for results i will tell you the method update Did not work with kernel remove_dgpu=0000:XX:XX.X

Any success u got???

Yassingames, 2024-07-23:
> > i think i found trying r 126 with the fix wait for results i will tell you the method update Did not work with kernel remove_dgpu=0000:XX:XX.X
> 
> Any success u got???

nope trying chromeos r121 now i am losing so much time trying to download these recovery images to test them so far r101 is the only version which kinda works and r101 and r113 are the only versions which install and run sucssesfully
[image]

Mir5382, 2024-07-23:
> nope trying chromeos r121 now i am losing so much time trying to download these recovery images to test them so far r101 is the only version which kinda works and r101 and r113 are the only versions [...]

Means 101 is the last option available...sadly

Mir5382, 2024-07-23:
> nope trying chromeos r121 now i am losing so much time trying to download these recovery images to test them so far r101 is the only version which kinda works and r101 and r113 are the only versions [...]

Tried 123 failed to boot....
Now trying 119 😀

Yassingames, 2024-07-23:
brother i will try ll      versions between r101 and r126 for you  iin the screenshot these are what i tested the fix n the screenshot are all the versions i tested =+ r121 tested now i will keep trying and keep you updated give me any contactmethod just in case

Mir5382, 2024-07-23:
> brother i will try ll versions between r101 and r126 for you in the screenshot these are what i tested the fix n the screenshot are all the versions i tested + r121 tested now i will keep trying and keep you updated give me any contact method just in case i was not online because i was so tired that i fell asleep in the balcony and the is 58 degree celcius i am felling so sick now i slept for 4 hours in the sun+ i tried r119 i tried r126 in an 11th gen laptop via usb it worked very fine seems that the problem might be with the dedicated nvidia graphics but send me a list of the versions you tried too

I think 126, 124, 123, 119... None booted

Yassingames, 2024-07-23:
read my list updated let us split everyone of try versions are you in a linux distro and give me a way to contact you

Yassingames, 2024-07-23:
> read my list updated let us split everyone of try versions are you in a linux distro and give me a way to contact you

+what is your avreage internetspeed and if you use windowss do you use wsl or linux mint usb i have many other questions that can help so give me a way to contact you telegram messanger discord is the best

Mir5382, 2024-07-23:
> read my list updated let us split everyone of try versions are you in a linux distro and give me a way to contact you

No brother...i don't use distro...but i use telegram...
Also i use wsl and without usb that is why it takes a lot of time for my PC to try one....
Also i am done for today😀
I will try tomorrow now....
Also internet speed here remains between 4 to 6 mbps

Yassingames, 2024-07-23:
> No brother...i don't use distro...but i use telegram... Also i use wsl and without usb that is why it takes a lot of time for my PC to try one.... Also i am done for today😀 I will try tomorrow [...]

ok local time for me it is 2 am i understand my download speeds avrege around 200kb and 1 mbs in best cases i am 5g but our laptops are old you know tommrow i will connect telegram to my pc and give you all the details my account is @DevHacker have a good night i will be awake to try r117 and r228 because if r118 works that will be so good thx see you

Yassingames, 2024-07-23:
forgot i use deepin 20.9 linux distro it is a bit faster than windows i install chromeos in my harddrive directly and linux just comes with grub so it is a bit easier than wsl

Mir5382, 2024-07-23:
> forgot i use deepin 20.9 linux distro it is a bit faster than windows i install chromeos in my harddrive directly and linux just comes with grub so it is a bit easier than wsl

Ohh.ok

Yassingames, 2024-07-23:
read the update and tell me what you think is not that hallirious

Yassingames, 2024-07-24:
r116 booted trying to get it  linux and playstore running wait for update and yesterday when you left i went to brunch discord the person who helps their refused to help saying he nows the fix but our devices are old i send him a message 10 lines long shaming him and told he is probably paid so he band me and said i am a bozo

Mir5382, 2024-07-24:
> r116 booted trying to get it linux and playstore running wait for update and yesterday when you left i went to brunch discord the person who helps their refused to help saying he nows the fix but [...]

Sure bro ... waiting....
Actually my eyes are burning very much so i am not using laptop yet 😀

Yassingames, 2024-07-24:
sorry do you use glasses wear them last week i had a 4hour minecraft session in the laptop without glasses when i went to the bathroom my eyes were so watery like i was crying i had a 48 hour headace and was not even able to read text of our 58 inch tv

Mir5382, 2024-07-24:
> crying

Same bro...i even got drops for it few months back but then i used phone very much and now it is very bad....i think i need to see doctor again

Yassingames, 2024-07-24:
> > crying
> 
> Same bro...i even got drops for it few months back but then i used phone very much and now it is very bad....i think i need to see doctor again

yeah i think you should and you should test your eyesight i see normally but the glasses are because of how much i use screens one eye is turning up and the other left and getting small in size

Yassingames, 2024-07-24:
hmm i kina stole a 4tb harddrive from my dad to store chromeos inbecause i am about to run out of space from the amount of versions i tested
i am now gonna test r116 and r120n r116 booted but i have trouble with my google account

Mir5382, 2024-07-24:
> yeah i think you should and you should test your eyesight i see normally but the glasses are because of how much i use screens one eye is turning up and the other left and getting small in size

Nah eyesight is normal.. doctor saying allergy....and use uv protected eyeglasses

Mir5382, 2024-07-24:
> hmm i kina stole a 4tb harddrive from my dad to store chromeos inbecause i am about to run out of space from the amount of versions i tested i am now gonna test r116 and r120n r116 booted but i have trouble with my google account

U don't delete those u try???

Yassingames, 2024-07-24:
> Nah eyesight is normal.. doctor saying allergy....and use uv protected eyeglasses

my glasse are uv protected only

Mir5382, 2024-07-24:
> > > > > crying
> > > > 
> > > > 
> > > > Same bro...i even got drops for it few months back but then i used phone very much and now it is very bad....i think i need to see doctor again
> > > 
> > > 
> > > yeah i think you should and you should test your eyesight i see normally but the glasses are because of how much i use screens one eye is turning up and the other left and getting small in size
> > 
> > 
> > Nah eyesight is normal.. doctor saying allergy....and use uv protected eyeglasses
> 
> my glasse are uv protected only allergy from what spring and dry eyes

Good then...
Should i try 116? 🤔 Or wait

Yassingames, 2024-07-24:
well nope because neither linux or google playstore works linux installtion break for some unknown reason i will try other versions and come back now i will try r118 with brunch r 118 not the i ran the r116 with brunch r126 unstable
i am gonna try r118 with brunch r118 stable and ext2 instead of ext4 hope that works wish me luck see you

Mir5382, 2024-07-24:
> well nope because neither linux or google playstore works linux installtion break for some unknown reason i will try other versions and come back now i will try r118 with brunch r 118 not the i ran [...]

Ohh ok best of the luck...btw i am installing 116 right now with r116 brunch

Yassingames, 2024-07-24:
ok wish you luck

Yassingames, 2024-07-24:
i am tryin r118 with brunch r118 and i am gonna try r120 with brunch r120 with ext2 instead of ext4

Mir5382, 2024-07-24:
> i am tryin r118 with brunch r118 and i am gonna try r120 with brunch r120 with ext2 instead of ext4

Hope it works

Mir5382, 2024-07-24:
> r116 booted trying to get it linux and playstore running wait for update and yesterday when you left i went to brunch discord the person who helps their refused to help saying he nows the fix but [...]

Yes bro same.m.. Everything is working except play services... Otherwise account got logged in too but sadly no play services

Mir5382, 2024-07-24:
> Yes bro same.m.. Everything is working except play services... Otherwise account got logged in too but sadly no play services

Actually... Apps are installed too but play services are not updating..... It is not as good as 101

Yassingames, 2024-07-25:
ok sorry for not updating for the last 30 hours but my laptop screen decided to break for no reason only the backlight worksbut 30 minutes aftewr my lasty post hopefully today our final package arrived we were moving from one city to another and i had a dell secondary monitor which arrived but it worked on vga so i had to wait for the vga cable and the hdmi to vga to arrive finaaly today 2hours ago it arrived and i finally was able to open the laptop r118 did not work also r120 sadly i am gonna try r123

Yassingames, 2024-07-25:
[image]
all versions tried expect r113 most were by brunch r126 stable r117 was brunch r126 unstable and r101 was with brunch r101 and r118 with brunh r118 and r120 with brunch r120 so far r101 is the only fully working version

Yassingames, 2024-07-25:
i am gonna test r119 now if it works it is a miracle hope tat happens i am now having 123 issuies with my life this week so do not be suprised if i just force delete myself

Yassingames, 2024-07-26:
r119 and r120 did not work did you get something working

Yassingames, 2024-07-26:
i just filledup my harddrive with 200gb worth of chromeos recovery files i am gonna test r123 and chromeos flex r126 and try to copy kernel and boot loader to chromeos r126 and if it failed iam gonna edit fydeos to work with us

Mir5382, 2024-07-26:
> i just filledup my harddrive with 200gb worth of chromeos recovery files i am gonna test r123 and chromeos flex r126 and try to copy kernel and boot loader to chromeos r126 and if it failed iam gonna edit fydeos to work with us

Actually after u told me that r116 is working so i tried that and somehow succeeded to install Android apps in it too but what i found with chrome os is that it has video playback issues with many apps like rvx YouTube and hotstar ...
So i tried fydeos 17 after that and although it installs and is better than chrome os infact but still i feel it lagging in my pc alongwith low volume and no display color changing settings...
So i then threw this idea out and stayed with windows 😥

Yassingames, 2024-07-26:
ok i tried now chromeos flex and fydeos well fyde os was hell bad so i threw out of the window and decided to instaal the bootloader in flex and try to put it in chromos

Yassingames, 2024-07-26:
see i noticed that in fydeos the gapps configartor is just a csv chrome extension so if i was able to get my hands on it and the fydeos substem for android i would be able to android support for chromeos flex which is plan a if failed i will try ro see the diffrence bettwen chromeos and chromeos flex recovery files swap and try to make chromeos work somehow for now here is my telegram user @DevHackerG add me because the reast of the plan is against github because chromeos is not open source and i can be charged for piracy

Mir5382, 2024-07-26:
> see i noticed that in fydeos the gapps configartor is just a csv chrome extension so if i was able to get my hands on it and the fydeos substem for android i would be able to android support for [...]

Ok btw i messaged on telegram... Don't know if it is u or someone else 😀

Mir5382, 2024-07-26:
> ok my phone is locked now so i can see you on telegram + i might chang the user name to @DevHacker check if my profile han anonyoums logo in it if not i changed it to @DevHacker but i changed plan to b i will copy efi and kernel from flex chromeos

Yes i have sent a message there

[1 more reply]