sebanc/brunch #2110: no sound at asus c302 while using newer kernel(>4.19), or screen rotation issue at chromebook-4.19

huLter, 2024-07-21:
now I am using 'chromebook-4.19' kernel with 'chromebook_audio' option, ChromeOS 126 image + Brunch r126 framework. everything works fine except tablet mode / screen auto rotation. it turns into tablet mode unless  it is completely folded back. if it is in tent mode / stand mode, nothing changes.
so I try to switch to other kernel. first I try 'chromebook-6.6' with 'chromebook_audio' option. tablet mode / screen auto rotation work perfectly but it has no sound. the volume bar is grey and there is no speaker / microphone in the setting page.
then I try regular '6.6' kernel with 'chromebook_audio' option. tablet mode / screen auto rotation still work. the volume bar get color, volume can be changed by key and mouse.  speaker / microphone appear in the setting page. but it still has no sound even i turn volume to max.
so i go back to 'chromebook-4.19' kernel. did anybody know how to fix this problem?

chuckiepatt, 2024-12-15:
I got the same chromebook kernel 4.19 doesn't work on Android 13 so your gonna have to roll back to android 12 for 4.19 to work so newer brunch framework 130 doesn't work for us.

xocox, 2025-05-24:
#2231 everything works in Asus C302