sebanc/brunch #2116: ChromeOS stuck at ChromeOS logo screen and brunch bootsplash not showing

Haseeb0309, 2024-08-02:
Hi sebanc 
I have Intel pentium g850 @2.90 GHz x2 
Gpu is mesa Intel hd graphics 2000 1.5 gb 
Since I installed dualboot  ChromeOS from linux after that brunch bootsplash not showing then ChromeOS logo screen appears and then stuck plz help

Haseeb0309, 2024-08-02:
Using rammus r126 with brunch 126

> Hi sebanc I have Intel pentium g850 @2.90 GHz x2 Gpu is mesa Intel hd graphics 2000 1.5 gb Since I installed dualboot ChromeOS from linux after that brunch bootsplash not showing then ChromeOS logo screen appears and then stuck plz help

darklord969, 2024-08-03:
hi i have same problem 
i have i5-12400 
GPU 5700xt rayzrn
Since I installed dualboot ChromeOS from windose after that brunch then ChromeOS logo screen appears and then stuck plz help

Haseeb0309, 2024-08-03:
Yes my friend can you change your brunch into verbose mode and can see those errors and tell me
1st error is ChromeOS startup: unable to send user display 
2nd is monitor not found
3rd id security policy not found 
 plz can you confirm these errors in your brunch