sebanc/brunch #2132: Chrome OS stuck on chrome logo

RamRamAndroid, 2024-09-05:
hi, now i have a sony vaio laptop with a nvidia geforce 310m dgpu.the problem is that its stuck on the chrome logo. but i can still access the tty. but when i try to go back it just freezes.i also ran dmesg.
 thank you in advance

pics: 1.https://github.com/user-attachments/assets/cf0e8554-4bd2-4449-a8c3-a5646d6adddb
        2.https://github.com/user-attachments/assets/d07b4874-65ab-459b-bf3b-8c628a147448

RamRamAndroid, 2024-09-05:
x

Chrome-God, 2024-09-30:
Sure! Here's a very simple answer focusing on your issue with Chrome:

1. **Check for Errors**:
   - In the TTY, type:
     ```bash
     dmesg | grep chrome
     ```

2. **Reboot**:
   - Try rebooting using this command:
     ```bash
     sudo reboot
     ```

3. **Fix Chrome Issue**:
   - If Chrome still freezes, try reinstalling [Chrome:](https://chrome.xahuapu.net/)
     ```bash
     sudo apt-get remove google-chrome-stable
     sudo apt-get install google-chrome-stable
     ```

4. **Graphics Driver**:
   - If it seems related to graphics (like the NVIDIA GPU), switch drivers:
     ```bash
     sudo apt-get install nvidia-driver-390
     sudo reboot
     ```

This should help with the Chrome freezing issue!