sebanc/brunch #2136: May I ask why I can not download kernel 5.15 with the newest brunch unstable ?

mirek190, 2024-09-18:
May I ask why I can not download kernel 5.15 with the newest brunch unstable?

My laptop is only works properly with that kernel .... 

Thanks

sebanc, 2024-09-18:
Standard Kernel 5.15 was not working properly with android apps on recent ChromeOS releases due to a missing kernel feature but you can still use the "chromebook-5.15" kernel.

mirek190, 2024-09-18:
What is a difference between normal kernel 5.15 and   chromebook-5.15?

sebanc, 2024-09-18:
The chomebook-5.15 kernel uses chromiumos kernel source whereas standard 5.15 kernel used mainline.

mirek190, 2024-09-18:
> The chomebook-5.15 kernel uses chromiumos kernel source whereas standard 5.15 kernel used mainline.

That is not explain anything   ;)

Are bigger differences like  chromiumos  5.15 kernel has cut off drivers or something?

sebanc, 2024-09-18:
It uses ChromeOS-Flex config. If you need a specific config and that it is missing you can post it here.

mirek190, 2024-09-18:
My device has a very specific es8336 driver ... Any official driver es8336 is not working.

I have a source of that driver only for kernel 5.12  but also working ok with 5.15 but is not compatible with higher kernels ( too many changes  ) 

I tried integrate the driver with chromium 5.15 and is giving errors during building ... different members of structs or one function has too many arguments.

With vanilla 5.15 is can build that but with chromium 5.15 not ... 🫥

mirek190, 2024-09-19:
> Standard Kernel 5.15 was not working properly with android apps on recent ChromeOS releases due to a missing kernel feature but you can still use the "chromebook-5.15" kernel.

What "missing kernel feature" ?
Maybe I will backport it from chromium kernel 5.15 to vanilla kernel 5.15?

What do you say ?

mirek190, 2024-10-17:
I just want inform you I fully ported driver to kernel 6.1  ;)

closed 2024-10-21 by mirek190