sebanc/brunch #2143: Need help installing ChromeOS & which recovery image to use?

skalwaghe-56, 2024-09-30:
Hi,

I am trying to install ChromeOS on my HDD while I will have windows on my SSD. I have created a 100GB NTFS drive for it, and installed the bobba recovery image. But when I run the chromeos-install.sh in my WSL i get the error. SS attached.

skalwaghe-56, 2024-10-01:
@sebanc I get the error Primary GPT header is invalid ......... While creating the images from my WSL onto my 100 GB HDD Partition

skalwaghe-56, 2024-10-01:
@sebanc please help me bro

skalwaghe-56, 2024-10-03:
Hey, @sebanc I managed to boot into brunch using grub2win after the first setup the ChromeOS logo appeared and the screen was stuck onto it. I tried installing the same way as mentioned in the guide. I am using shyvana for my Intel Pentium N3710 and 4gb Ram please help me out.

skalwaghe-56, 2024-10-03:
Currently I am trying to install through usb let's see

Abd-ulRahman, 2024-12-28:
You may be installing the wrong image I did have this issue for the last 4 days
You should find an old image that correspond your system as I did find mine at https://github.com/sebanc/brunch/issues/2001#issue-2092460378
Furthermore, the image you are installing must be suitable to your machine (Intell or AMD) And which generation is it.