sebanc/brunch #2154: Coral based boards are perfect match, babytiger boots perfectly, except no android even with fix. Anything I can try?

madhits, 2024-10-25:
So I have been on a rammus 116 for a while with everything working. I tried rammus 117 and got no boot. I assume higher would yield the same results. I looked again at coral/babytiger is identical to my laptop except ddr4 vs ddr3. Babytiger boots and runs great, even better then rammus 116. However I've got the play store error 7, yes I tried the android fix, no go.

Is there something else I can try? should I bother with unsupported brunch with coral?

madhits, 2024-10-29:
So it turns out it was a kernel 6x issue. I can use kernel 5.15 and I get arc++ in coral. So is arcvm kernel specific? Would I get arcvm from coral or is it just not available? I can boot kernel 6.15 in coral but playstore wont start. 

So question is really 2 part. 1. Does r126 have arcvm under coral?  2. If I'm able to boot kernel 6x, why wont android start?

Abd-ulRahman, 2024-12-28:
Try to add the following patch android_init_fix2 and see if it works
As android_init_fix2 Works especially with (Something went wrong. Error code: 8) and sometime with Error Code: 7

madhits, 2024-12-28:
I'll try that but do you know if arcvm is even available in coral r126? Kinda pointless to try higher kernel if it's not. I'm on kernel 5.15 now with arc++ in coral r126

Abd-ulRahman, 2024-12-28:
> I'll try that but do you know if arcvm is even available in coral r126? Kinda pointless to try higher kernel if it's not. I'm on kernel 5.15 now with arc++ in coral r126

I cannot ensure you that I am just new Working in that for the last 5 days
I did have many issues Installing multi core learned a lot seeing some youtube videos
In conclusion If you stuck in some image, I recommend you not to change or upgrade, Cause most of the times it won't work, and you'll get many issues.
Hope that helps you......!

closed 2025-11-29 by github-actions[bot]