sebanc/brunch #2159: Surface Pro wifi

Memowser, 2024-11-08:
Hey guys I recently learned about brunchos and started to do everything following the GiTHUB. However im running into issues where on the start screen i cant connect to the WIFI so i cant proceed with the next steps. Is this because of the brunchOS that i have or the ChromeOS or kernal??? Sorry im really new to all of this so if you can stupify it i would appreciate it! I'm trying to run it on my surface pro with a intel i5 7300u processor. I'm pretty sure its a 7th gen intel processor which I know it doesnt officially support it but from what i researched people have done it with no issues? I used the latest version of shyvana and the latest version of Brunchos. Thanks!

thadrumr, 2024-11-14:
I have the same generation Surface Pro 5 except mine is the Core M3 version.  I am also having issues with some Wireless Networks.  See Issue 2151

dshizntpdt, 2024-12-18:
So with 6th and 7th gen processors ive switched to [Brunch r126](https://github.com/sebanc/brunch/releases/tag/r126-stable-20240630) and "eve" 126 as that is based on the pixelbook which is 7th gen and support has been extended to 2027.  I've had really good luck with that combination.

thadrumr, 2024-12-18:
Are you staying on version 126 of chrome os?

thadrumr, 2024-12-18:
I also don’t think it’s an issue with processor support. This looks like an issue with Wi-Fi drivers.

thadrumr, 2025-01-10:
The people over at Fyde were able to fix this issue with version 19 SP1.  It would be really nice to have this fixed with Brunch as I really don't like paying for Fyde.

closed 2025-11-27 by github-actions[bot]