sebanc/brunch #2169: How to dual-boot without Grub2Win?

Andrew-Sor, 2024-12-14:
I can't use Grub2Win because it is no longer available in Russia. Are there any other ways to install Chrome OS with dual-boot without Grub2Win?

venterum, 2024-12-14:
You can use [grub2-brunch](https://github.com/M-L-P/grub2-brunch) as a boot option in a regular GRUB loader

Andrew-Sor, 2024-12-16:
> You can use [grub2-brunch](https://github.com/M-L-P/grub2-brunch) as a boot option in a regular GRUB loader

What do I need to do for this?

venterum, 2024-12-16:
What's your main system?

Andrew-Sor, 2024-12-16:
> What's your main system?

Windows 11 24H2

venterum, 2024-12-16:
Great, in that case you will need to install the GRUB bootloader. If you don't have linux, you can try installing refind bootloader into EFI partition, add it as a bootoption, and then put grub2brunch into boot folder. Using refind, boot into grub2brunch and then specify the path to the ChromeOS image and generated GRUB configuration in the settings

Andrew-Sor, 2024-12-17:
> Если вам тяжело общаться на английском, то можем перейти на русский язык.

Хорошо. А можно поподробнее о том как установить rEFInd?

Andrew-Sor, 2024-12-17:
> В интернете достаточно гайдов о том, как его установить, в том числе на базе винды. Думаю, что после небольшого ресёрча всё станет понятно)
> 
> От себя скажу, что для получения доступа к EFI разделу из винды можно смонтировать его, после чего использовать Explorer++ для манипуляций с файлами.

Хорошо, я поищу

Andrew-Sor, 2024-12-18:
> В интернете достаточно гайдов о том, как его установить, в том числе на базе винды. Думаю, что после небольшого ресёрча всё станет понятно)

Так, я установил rEFInd. Что теперь делать с grub2brunch?

Andrew-Sor, 2024-12-18:
Ладно, я разобрался с grub2brunch и попытался загрузить ChromeOS. Он всё зависло на логотипе brunch и надписи "Loading Brunch Framework..."

Что это может значить?

venterum, 2024-12-18:
> Он всё зависло на логотипе brunch и надписи "Loading Brunch Framework..."

Что у вас по железу? Brunch штука капризная, работает не везде. Больше всего интересует процессор(в частности его поколение), а также видеокарта.

Andrew-Sor, 2024-12-18:
> > Он всё зависло на логотипе brunch и надписи "Loading Brunch Framework..."
> 
> Что у вас по железу? Brunch штука капризная, работает не везде. Больше всего интересует процессор(в частности его поколение), а также видеокарта.

[image]
Видеокарты нету

venterum, 2024-12-18:
А какой вы образ собирали? Voxel?

Andrew-Sor, 2024-12-19:
> А какой вы образ использовали? Voxel?

Да

Andrew-Sor, 2024-12-19:
> Voxel?

Да, я использовал образ Voxel.

venterum, 2024-12-20:
> > Voxel?
> 
> Да, я использовал образ Voxel.

Хмм, в таком случае, боюсь что ничем не смогу помочь, все проблемы с бранчем очень индивидуальны т.к. его работа очень зависит от железа. Если есть желание проверить, в бранче проблема, или это просто ChromeOS не хочет запускаться у вас на ПК, то попробуйте сделать загрузочную флешку с ChromeOS Flex. Если она запустится без проблем, то, возможно, что-то получится сделать.

Желаю удачи в устранении проблемы!

Andrew-Sor, 2024-12-20:
> Хмм, в таком случае, боюсь что ничем не смогу помочь, все проблемы с бранчем очень индивидуальны т.к. его работа очень зависит от железа. Если есть желание проверить, в бранче проблема, или это [...]

Ладно, попробую с флешки. Спасибо за помощь

Andrew-Sor, 2024-12-21:
> загрузочную флешку с ChromeOS Flex. Если она запустится без проблем, то, возможно, что-то получится сделать.

С флешки Chrome OS загрузился нормально. Но она очень медленная и почему-то он застрял на попытке войти в аккаунт.

Andrew-Sor, 2024-12-21:
И я записывал не Flex, а тот образ который использовал в grub2brunch

Andrew-Sor, 2024-12-23:
> Хмм, в таком случае, боюсь что ничем не смогу помочь, все проблемы с бранчем очень индивидуальны т.к. его работа очень зависит от железа. Если есть желание проверить, в бранче проблема, или это [...]

Кажется проблема в grub2brunch. Можете ли вы подсказать ещё альтернативы Grub2Win?

Andrew-Sor, 2024-12-24:
Я нашёл решение! Я использовал загрузчик из сгенерированного img, изменив файл grub.conf.

Всё, можно закрывать issue

closed 2024-12-24 by Andrew-Sor

ioc2e3, 2025-08-14:
> Great, in that case you will need to install the GRUB bootloader. If you don't have linux, you can try installing refind bootloader into EFI partition, add it as a bootoption, and then put grub2brunch into boot folder. Using refind, boot into grub2brunch and then specify the path to the ChromeOS image and generated GRUB configuration in the settings.
> 
> Если вам тяжело общаться на английском, то можем перейти на русский язык.

Excuse me ...

On hard drive where Windows 10 is installed ...

No Grub2Win ...

How to dual-boot Windows and ChromeOS ?

.

Andrew-Sor, 2025-08-19:
> Excuse me ...
>
> On hard drive where Windows 10 is installed ...
>
> No Grub2Win ...
>
> How to dual-boot Windows and ChromeOS ?
>
> .

Why can't you use Grub2Win? If there are good reasons for this, then I can tell you how I did it.

Andrew-Sor, 2025-08-19:
> > Why can't you use Grub2Win? If there are good reasons for this, then I can tell you how I did it.
> 
> That ...
> 
> Grub2Win doesn't have Traditional Chinese user interface ...
> 
> .

How do you write in English? With a translator? I think it would be easier to use a translator as well. My method requires access to the EFI system partition. It is complicated.

Andrew-Sor, 2025-09-07:
> let me know the solution you used, i have the same problem as you, can't download Grub2win. thank you

I used the bootloader from the ChromeOS image with the generated config file. For dual-boot I used Refind.

Karen3251, 2025-09-09:
Yes, I'm using an Intel 11th gen (i5 1135g7), i have WSL installed wtih unbuntu and I disabled scureboot.
Will it be helpful for you?

Andrew-Sor, 2025-09-09:
> Yes, I'm using an Intel 11th gen (i5 1135g7), i have WSL installed wtih unbuntu and I disabled scureboot. Will it be helpful for you?

Great, then I wrote a short instruction:

<details>
  <summary>Click to open guide</summary>

# Installing Dual-Boot without Grub2Win

### Requirements:
* Administrator access.
* Prepared NTFS disk partition of at least 16 GB, unencrypted (BitLocker disabled).
* Installed WSL with any Linux distribution.
* A computer compatible with running Brunch.
* Basic knowledge of the Linux terminal.
* A program for reading `.img` files.
* Access to the system EFI partition.
* Disabled Secure Boot.

### Rolling up

1. Download the latest Brunch release and unzip it to a convenient location.

2. Download the recovery file that suits your processor:
    * Intel:
        * 8th gen & 9th gen: "[shyvana](https://cros.tech/device/shyvana)" for Intel / "[bobba](https://cros.tech/device/bobba)" for Celeron.
        * 10th gen: "[jinlon](https://cros.tech/device/jinlon)".
        * 11th gen & above: "[voxel](https://cros.tech/device/voxel/)".
    * Ryzen:
        * "[gumboz](https://cros.tech/device/gumboz/)".

3. Place the recovery file from the archive in the same folder and rename it to `recovery.bin`.

4. Launch the terminal of your Linux distribution and go to the folder with the files.

5. Run the command `sudo bash chromeos-install.sh -src chromeos_filename.bin -dst [path to the prepared disk]/chromeos.img -s size`
    * It is best to insert the path to create the image like this:
        1. Create any file in the desired location and name it `chromeos.img`.
        2. Move the cursor after `-dst` and move this file to the terminal.
    * Instead of `size`, you can specify the desired image size (in GB). The default is 14 GB.

6. When the installer asks you about the installation type, type `dualboot` in the terminal and press Enter.

7. Close the terminal and go to the source files.

8. Extract the `/efi/boot` folder from `efi_legacy.img` and rename it to `ChromeOS`.

9. Replace the contents of the `grub.cfg` file with the contents of the generated configuration file (it is located in the same place as the built image).

10. Move the bootloader folder to the system EFI partition.

11. Install Refind or another bootloader manager, following the instructions on the Internet. To boot ChromeOS, use `bootia64.efi` or `bootia32.efi`.
</details>

*Note:*
* *I used a translator when writing this guide, so if you don't understand something, please ask.*
* *This guide may contain inaccuracies because it's been a long time since I installed ChromeOS. If you have any problems or difficulties, please ask too.*

Karen3251, 2025-09-27:
OMG, it works. Thank you so muchhh