sebanc/brunch #2173: [Compatibility] Intel Core-i5 3rd Gen running r119 working well

Abd-ulRahman, 2024-12-28:
According to the previous issue by [signigelchan](https://github.com/signigelchan)
[Nigel Chan](https://github.com/signigelchan)

[Compatibility] Intel Core-i5 3rd Gen (Ivy Bridge) running r119 working well #2001

And after testing 

Google Play worked after modifying android_init_fix2 on framework kernel-5.15
Please add android_init_fix2 to the new kernels that would make Google Play works After patching for people how have hard time to make it work

 Hardware Spec
|  CPU | 
|------------------------------------------|
|  Dell Latitude E6330 Core-i5 3rd Gen | 

Thanks in advance
Best Regards

Baretta89, 2025-02-09:
Sorry i dont understand 
Can ypu explain more how to add android_init_fix2

Abd-ulRahman, 2025-02-09:
As in brunch-setup for [Brunch r111 stable 20230414](https://github.com/sebanc/brunch/releases/tag/r111-stable-20230414) it was there

	case "$selected_kernel" in
		6.1)
	options_page1="enable_updates;pwa;android_init_fix;android_init_fix2;mount_internal_drives;disable_sdcard_storage;disable_usb_storage;chromebook_audio;native_chromebook_image;broadcom_wl;rtl8188eu;rtl8192eu;rtl8723bs;rtl8723bu;rtl8723du;rtl8812au;rtl8814au;rtl8821ce;rtl8821cu;rtl88x2bu;rtl8852ae"
	options_page2="ipts_touchscreen;ithc_touchscreen;no_camera_config;invert_camera_order;acpi_power_button;alt_touchpad_config;alt_touchpad_config2;alt_touchpad_config3;touchpad_resume_fix;internal_mic_fix;internal_mic_fix2;sysfs_tablet_mode;force_tablet_mode;suspend_s3;advanced_als"
		;;
		5.15)
	options_page1="enable_updates;pwa;android_init_fix;android_init_fix2;mount_internal_drives;disable_sdcard_storage;disable_usb_storage;chromebook_audio;native_chromebook_image;broadcom_wl;iwlwifi_backport;rtl8188eu;rtl8188fu;rtl8192eu;rtl8723bu;rtl8723du;rtl8812au;rtl8814au;rtl8821ce;rtl8821cu;rtl88x2bu;rtl8852ae;rtbth"
	options_page2="ipts_touchscreen;ithc_touchscreen;no_camera_config;invert_camera_order;acpi_power_button;alt_touchpad_config;alt_touchpad_config2;alt_touchpad_config3;touchpad_resume_fix;internal_mic_fix;internal_mic_fix2;sysfs_tablet_mode;force_tablet_mode;suspend_s3;advanced_als"
		;;
		5.10)
	options_page1="enable_updates;pwa;android_init_fix;android_init_fix2;mount_internal_drives;disable_sdcard_storage;disable_usb_storage;chromebook_audio;native_chromebook_image;broadcom_wl;iwlwifi_backport;rtl8188eu;rtl8188fu;rtl8192eu;rtl8723bu;rtl8723de;rtl8723du;rtl8812au;rtl8814au;rtl8821ce;rtl8821cu;rtl88x2bu;rtl8852ae;rtbth"
	options_page2="ipts_touchscreen;ithc_touchscreen;goodix;no_camera_config;invert_camera_order;acpi_power_button;alt_touchpad_config;alt_touchpad_config2;alt_touchpad_config3;touchpad_resume_fix;internal_mic_fix;internal_mic_fix2;sysfs_tablet_mode;force_tablet_mode;suspend_s3;advanced_als"
		;;
		5.4)
	options_page1="enable_updates;pwa;android_init_fix;android_init_fix2;mount_internal_drives;disable_sdcard_storage;disable_usb_storage;chromebook_audio;native_chromebook_image;broadcom_wl;iwlwifi_backport;rtl8188eu;rtl8188fu;rtl8192eu;rtl8723bu;rtl8723de;rtl8723du;rtl8812au;rtl8814au;rtl8821ce;rtl8821cu;rtl88x2bu;rtbth"
	options_page2="ipts_touchscreen;goodix;no_camera_config;invert_camera_order;acpi_power_button;alt_touchpad_config;alt_touchpad_config2;alt_touchpad_config3;touchpad_resume_fix;internal_mic_fix;internal_mic_fix2;sysfs_tablet_mode;force_tablet_mode;suspend_s3;advanced_als"
		;;
		4.19) options_page1="enable_updates;pwa;android_init_fix;android_init_fix2;mount_internal_drives;disable_sdcard_storage;disable_usb_storage;chromebook_audio;native_chromebook_image;broadcom_wl;iwlwifi_backport;rtl8188eu;rtl8188fu;rtl8192eu;rtl8723bu;rtl8723de;rtl8723du;rtl8812au;rtl8814au;rtl8821ce;rtl8821cu;rtl88x2bu;rtbth"
	options_page2="goodix;no_camera_config;invert_camera_order;acpi_power_button;alt_touchpad_config;alt_touchpad_config2;alt_touchpad_config3;touchpad_resume_fix;internal_mic_fix;internal_mic_fix2;sysfs_tablet_mode;force_tablet_mode;suspend_s3;advanced_als"
		;;
		*)
	options_page1="enable_updates;pwa;android_init_fix;android_init_fix2;mount_internal_drives;disable_sdcard_storage;disable_usb_storage;chromebook_audio;native_chromebook_image;no_camera_config;invert_camera_order;acpi_power_button;alt_touchpad_config;alt_touchpad_config2;alt_touchpad_config3;touchpad_resume_fix;internal_mic_fix;internal_mic_fix2;sysfs_tablet_mode;force_tablet_mode;suspend_s3;advanced_als"
		;;
	esac

Abd-ulRahman, 2025-02-09:
and it's code was

	fi

	if [ "$android_init_fix2" -eq 1 ]; then
		if [ -f /roota/etc/init/arc-ureadahead.conf ]; then rm /roota/etc/init/arc-ureadahead.conf; fi
		if [ -f /roota/etc/init/arc-lifetime.conf ]; then rm /roota/etc/init/arc-lifetime.conf; fi
		if [ -f /roota/etc/init/arc-keymasterd.conf ]; then rm /roota/etc/init/arc-keymasterd.conf; fi
		sed -i "s@start on continue-arc-boot@start on continue-arc-boot and stopped arc-init-wait@g" /roota/etc/init/*
		if [ ! "$?" -eq 0 ]; then ret=$((ret + (2 ** 0))); fi
		cat >/roota/etc/init/arc-init-wait.conf <<INIT
	start on stopped startup
	script
		arc-init-wait
	end script

closed 2025-11-25 by github-actions[bot]