sebanc/brunch #2175: Please Correct [Request]

Abd-ulRahman, 2025-01-01:
Can you please Give me some help with this List about Chromeos rev
 ## Chromeos stable Revisions

| Revision | Stable Build | Comment |
|---------|---------------|---------------------|
| 91 | 13904.77.0 | ✓ |
| 91-2 | 13904.97.0 | ✓ |
| 92 | 13982.82.0 | ✓ |
| 92-2 | 13982.88.0 | ✓ |
| 93 | 14092.66.0 | ✓ |
| 94 | 14150.74.0 | ✓ |
| 94-2 | 14150.87.0 | ✓ |
| 98 | 14388.61.0 | ✓ |
| 98-2 | 16388.62.1 | ✓ |
| 100 | 14526.69.0 | ✓ |
| 103-0 | 14816.99.0 | Could be not sure |
| 103 | 14816.131.0 | ✓ |
| 103-2 | 14816.131.5 | ✓ |
| 104-0 | 14909.132.0 | Could be not sure |
| 105-0 | 14989.86.0 | Could be not sure |
| 106-0 | 15054.115.0 | Could be not sure |
| 107-0 | 15117.112.0 | Could be not sure |
| 108-0 | 15183.59.0 | Could be not sure |
| 109-0 | 15236.66.0 | Could be not sure |
| 110 | 15278.64.0 | ✓ |
| 113-0 | 15393.48.0 | Could be not sure |
| 113 | 15393.58.0 | ✓ |
| 114-0 | 15437.61.0 | Could be not sure |
| 114 | 15437.42.0 | ✓ |
| 115-0 | 15474.70.0 | Could be not sure |
| 115 | 15474.84.0 | ✓ |
| 116-0 | 15509.72.0 | Could be not sure |
| 116 | 15509.81.0 | ✓ |
| 117 | 15572.63.0 | ✓ |
| 118 | 15604.57.0 | ✓ |
| 119 | 15633.72.0 | ✓ |
| 120 | 15662.76.0 | ✓ |
| 121 | 15699.72.0 | ✓ |
| 122 | 15699.72.0 | ✓ |
| 123 | 15786.58.0 | ✓ |
| 124 | 15823.60.0 | ✓ |
| 125 | 15853.67.0 | ✓ |
| 126 | 15886.44.0 | ✓ |
| 127 | 15917.71.0 | ✓ |
| 128 | 15964.59.0 | ✓ |
| 129 | 16002.60.0 | ✓ |
| 130 | 16033.58.0 | ✓ |

Thanks in Advance
Best Regards

Abd-ulRahman, 2025-01-02:
All data collected over the internet 
Please confirm the right data

sebanc, 2025-01-02:
Not sure what you want to do with this but you can find the information in the below github repository:
https://github.com/jay0lee/chromeos-update-directory

Abd-ulRahman, 2025-01-02:
> Not sure what you want to do with this but you can find the information in the below github repository: https://github.com/jay0lee/chromeos-update-directory

Thanks for your reply

closed 2025-11-25 by github-actions[bot]