sebanc/brunch #2178: Stuck at loading Chrome OS

BasGame1, 2025-01-13:
Hi, I tried to install a dual boot with windows and chrome OS on a Dell Latitude 3580 laptop (with a i5-7000k and a AMD graphics card) but it got stuck at "applying patches" in a few hours I tried to boot again, install the WSL another time but it failed again. So I tried to create a bootable USB with Rufus instead of using grub2win. But now it's stuck at "loading brunch framework" I'm new at this please help

Abd-ulRahman, 2025-01-15:
If you Stuck at loading Chrome OS or at applying patches or at "loading brunch framework" that mean Chrome OS does not match with your platform in other word the hardware of your laptop does not meet with Chrome OS try to search for the right Chrome OS that meet with your laptop
Hope that helps

BasGame1, 2025-01-19:
Thank you, I tried shyvana and know it passes that screen, but it keeps rebooting itself. But thanks a lot