sebanc/brunch #2190: ChromeOS does not boot after updating Brunch from r129 to r132 on ChromeOS r130

Anonyoumss, 2025-02-09:
**Hello,
I need help in this situation I use ChromeOS daily for my study and work and i need al my data,
I updated brunch from r129 to r132 using the new pwa dashboard while running chromeos r130 after updating.I restarted and I cannot boot to chromeos or brunch settings from grub and i get a weird cannot boot you need to load kernel first.I will post it in an update.I had been using it for 3 months an it was very stable till I updated.**
Here is my grub config.

echo
echo Entry   Chrome
echo
echo Imported On  Thursday  November 21, 2024  at  8:50:14 PM
echo From file    "D:\chromeos.img.grub.txt"
echo
        rmmod tpm
        img_path=/chromeos.img
        img_uuid=5dd20420-e014-446e-972a-0e029a6966a3
        search --no-floppy --set=root --file $img_path
        loopback loop $img_path
        source (loop,12)/efi/boot/settings.cfg
        if [ -z $verbose ] -o [ $verbose -eq 0 ]; then
                linux (loop,7)$kernel boot=local noresume noswap loglevel=7 options=$options chromeos_bootsplash=$chromeos_bootsplash $cmdline_params \
                        cros_secure cros_debug img_uuid=$img_uuid img_path=$img_path \
                        console= vt.global_cursor_default=0 brunch_bootsplash=$brunch_bootsplash quiet
        else
                linux (loop,7)$kernel boot=local noresume noswap loglevel=7 options=$options chromeos_bootsplash=$chromeos_bootsplash $cmdline_params \
                        cros_secure cros_debug img_uuid=$img_uuid img_path=$img_path
        fi
        initrd (loop,7)/lib/firmware/amd-ucode.img (loop,7)/lib/firmware/intel-ucode.img (loop,7)/initramfs.img
     echo '       ** Ready To Boot **'

Anonyoumss, 2025-02-09:
Also is there a method for me to get my data, and is there a method for me to make it a full disk installtion i have it in an ext4 partition Thanks a lot,

joeyoropesa-dev, 2025-03-12:
I've heard that Google slowly replacing ChromeOS linux kernels with android kernels and android components. I don't know but this problem could be related to this - I'm not sure that Brunch Framework supports booting Android kernels (yet)

Anonyoumss, 2025-03-14:
> I've heard that Google slowly replacing ChromeOS linux kernels with android kernels and android components. I don't know but this problem could be related to this - I'm not sure that Brunch Framework supports booting Android kernels (yet)

Ok maybe this True but I got the issue fixed brunch did not update properly so I booted a Linux distro mounted my chromes.img file extracted brunch and used  some coomand to swap root-c from my ChromeOS..img with the one I extracted and it worked thanks @sebanc for help and it was straight forward I then updated chrome os to r134 I think and it worked.
I will mark this as closed soon so if there is anything ask!