sebanc/brunch #2191: login issue when installing "Could not mount cryptohome"

AdeeGaming01, 2025-02-09:
I'm having issues using the USB to install onto an SSD, when I try logging in I get a Please wait... screen for a few minutes, after which there's a "Could not mount cryptohome" error. I've tried turning TPM on or off same results
Brunch and ChromeOS ver: 132
CPU: Ryzen 5 5600x
GPU: RX 7900XT
Kernel ver: 6.1
Mobo: MSI B450 Gaming Plus Max (I assume that's not relevant, but I've had instances where it was the culprit with some issue)
USB should be about 14.4 GB in size

[image]

nunolu, 2025-02-20:
Hi, I have the same issue with r132

laconeo, 2025-02-24:
The same happens to me with a lenovo ideapad slim 3 14anm8 con ryzen 3

RicardoRien, 2025-03-09:
I'm facing the same issue. Intel core i 7 8th Gen. Latest version of chromeOs and Brunch.

PeterDD9PS, 2025-05-15:
Try a different recovery file! With the recovery file “sentry” I had the same error. For my Lenovo Yoga 260, the recovery file “nami” works perfectly without any errors!!!

I initially used the recovery file “sentry” for a Lenovo ThinkPad Yoga 260 (Intel® Core™ i7-6600U, Intel® HD Graphics 520), because its hardware is very similar to the Lenovo Thinkpad 13 Chomebook. But I could not fix the error “Could not mount cryptohome” with this recovery file. Only a guest login works. The Google account can then be accessed with the Chrome browser. If you then log out and try to set up an account again, the error “Cyptohome corrupted” appears. There is no way to fix the error with “sentry”.

My solution:
I have tried several other recovery files. Of course, the recovery file “reven” works in combination with the brunch framework. The login also works here - even with the TPM 2.0 module deactivated. However, the PlayStore is missing in “reven”. You can also install the recovery file “reven” without the brunch framework. But with the framework, the root shell works via “CRTL-ALT-F2” and the developer mode also works.

For my Lenovo Yoga 260, the recovery file “nami” works perfectly!!! This recovery file also supports the hardware of my Yoga 260 and, unlike “sentry”, the error message did not appear, i.e. I was able to log in successfully:

[image]

dineshdhotrad, 2025-07-30:
<html>
<body>
<html><head></head><body><div class="relative"><div class="prose text-pretty dark:prose-invert inline leading-normal break-words min-w-0 [word-break:break-word]"><span class="mt-md block"></span>
<p class="my-0"><strong>✅  FIXED: “Could not mount cryptohome / login loop” on AMD Ryzen laptops</strong></p><span class="mt-md block"></span>
<p class="my-0">Hi everyone,</p><span class="mt-md block"></span>
<p class="my-0">I was hitting exactly the same symptoms described in this thread:</p>
<ul class="marker:text-textOff list-disc">
<li>
<p class="my-0">ChromeOS boots but stalls after signing in (or Guest mode never proceeds)</p>
</li>
<li>
<p class="my-0">Repeated “Could not mount cryptohome” errors in VT-2 log</p>
</li>
</ul>
<p class="my-0">Hardware that failed: <strong>Lenovo Legion 5 15ACH6H – Ryzen 7 5800H (Zen 3 + Vega 8)</strong></p><span class="mt-md block"></span>
<p class="my-0">After a lot of testing I found the root cause and a repeatable solution.</p>
<h2 class="mb-xs mt-5 text-base font-[500] first:mt-0 dark:font-[475]">Why it happens</h2>
<p class="my-0">ChromeOS recovery images are board-specific.<br>
Using an image built for the wrong SoC (e.g. <em>rammus</em>, <em>volteer</em>, <em>skyrim</em>) leaves AMDGPU and cryptohome in an undefined state, which triggers the login loop.</p>
<h2 class="mb-xs mt-5 text-base font-[500] first:mt-0 dark:font-[475]">Working solution</h2>
<ol class="marker:text-textOff list-decimal">
<li>
<p class="my-0"><strong>Match your CPU generation to a real Chromebook.</strong></p>
<ul class="marker:text-textOff list-disc">
<li>
<p class="my-0">Ryzen 5000-C series Chromebooks = <em>guybrush</em> board</p>
</li>
<li>
<p class="my-0">Example devices:<br>
– HP Pro c645 G2 Chromebook<br>
– Acer Chromebook Enterprise Spin 514 (CP514-3WH series)</p>
</li>
</ul>
</li>
<li>
<p class="my-0"><strong>Download the correct recovery image.</strong><br>
<code>https://cros.tech/board/guybrush/</code> → grab the latest stable ZIP and extract the <code>.bin</code>.</p>
</li>
<li>
<p class="my-0"><strong>Pair it with the latest Brunch (r125+).</strong><br>
Brunch r125 ships a 6.1 kernel that already supports Zen 3/Vega.</p>
</li>
<li>
<p class="my-0"><strong>Standard install procedure</strong> (USB test or <code>chromeos-install.sh</code> single-boot):</p>
<div class="w-full md:max-w-[90vw]"><pre class="not-prose w-full rounded font-mono text-sm font-extralight"><div class="codeWrapper text-textMainDark selection:text-super selection:bg-super/10 bg-offset my-md relative flex flex-col rounded font-mono text-sm font-thin"><div class="translate-y-xs -translate-x-xs bottom-xl mb-xl sticky top-0 flex h-0 items-start justify-end"><button data-testid="copy-code-button" type="button" class="focus-visible:bg-offsetPlus hover:bg-offsetPlus text-textOff  hover:text-textMain dark:hover:bg-offsetPlus dark:hover:text-textMainDark font-sans focus:outline-none outline-none outline-transparent transition duration-300 ease-out font-sans  select-none items-center relative group/button  justify-center text-center items-center rounded-full cursor-pointer active:scale-[0.97] active:duration-150 active:ease-outExpo origin-center whitespace-nowrap inline-flex text-sm h-8 aspect-square"><div class="flex items-center min-w-0 font-medium gap-1.5 justify-center"><div class="flex shrink-0 items-center justify-center size-4"><svg xmlns="http://www.w3.org/2000/svg" width="16" height="16" viewBox="0 0 24 24" fill="none" stroke="currentColor" stroke-width="1.7999999999999998" stroke-linecap="round" stroke-linejoin="round" class="tabler-icon tabler-icon-copy "><path d="M7 7m0 2.667a2.667 2.667 0 0 1 2.667 -2.667h8.666a2.667 2.667 0 0 1 2.667 2.667v8.666a2.667 2.667 0 0 1 -2.667 2.667h-8.666a2.667 2.667 0 0 1 -2.667 -2.667z"></path><path d="M4.012 16.737a2.005 2.005 0 0 1 -1.012 -1.737v-10c0 -1.1 .9 -2 2 -2h10c.75 0 1.158 .385 1.5 1"></path></svg></div></div></button></div><div class="-mt-xl"><div><div data-testid="code-language-indicator" class="text-text-200 bg-background-300 py-xs px-sm inline-block rounded-br rounded-tl-[3px] font-thin">bash</div></div><div class="pr-lg"><span style="font-size: inherit; font-family: inherit; background: transparent; color: rgb(197, 200, 198); border-radius: 3px; display: flex; line-height: 1.42857; overflow-x: auto; white-space: pre;"><code style="white-space: pre-wrap; font-size: inherit; font-family: inherit; line-height: 1.66667; padding: 8px;"><span style="opacity: 1; font-size: inherit; line-height: 1.42857; color: rgb(197, 200, 198); background-color: transparent; flex-shrink: 0; padding: 8px; text-align: right; user-select: none;"><span class="token token" style="color: rgb(197, 200, 198);">sudo</span><span> </span><span class="token token" style="color: rgb(197, 200, 198);">bash</span><span> chromeos-install.sh </span><span class="token token punctuation">\</span><span>
</span></span><span style="opacity: 1; font-size: inherit; line-height: 1.42857; color: rgb(197, 200, 198); background-color: transparent; flex-shrink: 0; padding: 8px; text-align: right; user-select: none;"><span>     -src chromeos_xxx_guybrush_recovery.bin </span><span class="token token punctuation">\</span><span>
</span></span><span style="opacity: 1; font-size: inherit; line-height: 1.42857; color: rgb(197, 200, 198); background-color: transparent; flex-shrink: 0; padding: 8px; text-align: right; user-select: none;"><span>     -dst /dev/sdX           </span><span class="token token" s [... cut ...]
<p class="my-0">Add a short <strong>“CPU → Recommended board”</strong> table to the Brunch README.</p>
</li>
<li>
<p class="my-0">Close this issue as <em>resolved</em> with the procedure above.</p>
</li>
</ul>
<p class="my-0">Hope this saves others hours of trial-and-error.<br>
Thanks to @sebanc and all contributors for keeping Brunch alive!</p><span class="mt-md block"></span>
<p class="my-0">— <em>Lenovo Legion 5 user</em></p></div></div></body></html>
</body>
</html>

[image]
[image]
[image]
[image]