sebanc/brunch #2192: Ryzen 7 4800H and ChromeOS via Brunch Framework Stuck at Boot Screen

linuxtopia, 2025-02-11:
Hello,

I have a laptop Acer Nitro 5 with Ryzen 7 4800H cpu, would like to try ChromeOS via brunch framework but could not succeed. Used balena etcher or usbimager kind of bootable usb creation tools, no chance.

What I have used to create image;

brunch_r132_stable_20250208
chromeos_16093.78.0_zork with gumboz also morpheus.

[image]

What should I do?

KANGKKKKKKK, 2025-02-14:
Please make sure you select the appropriate version

linuxtopia, 2025-02-14:
You meant under gumboz' section versions? If so how can identify the exact version? Brunch main page asks to use gumboz for Ryzen cpu/apu.

do you have any info/idea for Ryzen 7 4800H?