sebanc/brunch #2194: Brunch framework Stuck at "Patches are being applied", Please Help

AstroCipher, 2025-02-19:
I am using brunch on dual boot setup windows+chromeos. Everything was fine but today one of my friend who is more techy tried to upgrade it. We first enabled updates and PWA from brunch settings, then we first updated it to stable brunch 132 and then rebooted, it successfully got booted, then we download chromeos update (chromeos132volteer) from (chromeos130volteer), after downloading, it shows the prompt to restart, after restarting it is stucked at "patches are being applied", after 15-20 min I force restarted the laptop and changed the kernal to 6.6 (earlier it was 6.6 after update it was changed to 6.12, I manually changed to 6.6) but it is still stuck at the same screen. I have some very important files in it. Please help. (If it is not possible to restore this, then is there any way to extract these files, as I am on dual boot?? My laptop:- Acer Aspire 3 8/512 Intel i3 1215u

BasGame1, 2025-04-20:
I'm with the same problem with my laptop dell latitude 3580

BasGame1, 2025-10-20:
I fixed it by using an older version of brunch and firmware