sebanc/brunch #2202: Multi touch and pen function don't work after sleep mode on Surface 9 pro

Frilinsh, 2025-03-11:
I have Surface 9 pro. Before  tablet go to sleep Chrome os is working great. After device switch to sleep mode and I switch it on the  multi touch and pen function don't work, I need to restart system to bring back multi touch and pen working. 

I am on Brunch r133 stable 20250308 and chrome os 132.0.6834.190. Also I have dualboot with windows and ubuntu installed. 

I tried all kernels and options in Brunch Configuration Menu nothing work to resolve this issue.  Also I tried different chrome os and brunch versions, problem was on all recent versions.

Please help.

bobyenser, 2025-03-30:
Same issue here on a surface pro 6.  I'm on Brunch r134 20250322 and rammus chrome os 133.0.6943.184.  I've tried multiple kernels and options in brunch.  My config is not dualboot.

Wifi will periodically not function after sleep also.  Wifi is enabled but it will not show any wifi networks.  Toggle wifi on/off does nothing.  Only way to fix these issues is to restart the machine.

mq20050522, 2025-04-23:
Same issue here on a surface pro 8. Multi touch and pen function stopped working after sleep mode and I need to reboot the system to bring back multi touch and pen function. I'm on Brunch Brunch r135 20250421 and voxel chrome os 135.0.7049.104.

Paulicat, 2025-08-29:
Fixed this issue with the following:
**_### Create file: "sudo nano /etc/init/iptsd-resume.conf" with contents:_**
`description "Restart iptsd on resume"

start on screen-locked

task

script
    echo "[iptsd-resume] running at $(date)" >> /var/log/iptsd-resume.log
    /usr/bin/restart-iptsd.sh >> /var/log/iptsd-resume.log 2>&1
end script`

### **_Create file: "sudo nano /usr/bin/restart-iptsd.sh" with contents:_**
`#!/bin/sh
set -e

sleep 1

# Kill any running iptsd
pkill -x iptsd || true

IPTS_MODULE=/lib/modules/6.12.15-generic-brunch-sebanc/ipts.ko

# Reload ipts kernel module
if lsmod | grep -q '^ipts'; then
    rmmod ipts 2>/dev/null || true
fi

if [ -f "$IPTS_MODULE" ]; then
    insmod "$IPTS_MODULE"
fi

sleep 0.5

# Start iptsd on the detected device
exec /usr/bin/iptsd "$(iptsd-find-hidraw)" &`

then sudo chmod +x /usr/bin/restart-iptsd.sh

**Special Note: locate on your system where the ipts.ko module is. This is taken from R138 running the 6.12.15 kernel. It will be in a different location if you are running a different kernel.**

This script kills any running iptsd, unloads the ipts module, reloads the ipts module, rebinds the iptsd after the ipts module is reloaded since the hidraw device often changes after a sleep/wake cycle. This all happens when you wake up the device and the lock screen is shown. Upstart in ChromeOS doesn't emit the "resume" notifier that would normally be emitted when a device is woken from sleep so the best I could find is to run this when the lock screen is displayed. If anyone has a better idea on what to attach the Upstart conf to please share. Also, the wifi wake issue (wifi dies after sleep/wake cycle) can be fixed by grabbing the mwifiex driver code from linux-surface 6.14.3 and building your own brunch release with the replaced driver source. Sebanc has been given all the above info so hopefully it can make it into the next brunch release.