sebanc/brunch #2212: Install Steam

SteveZMTstudios, 2025-04-05:
I cannot use `install_coin` crosh command nor [chromeos-steam://install](chromeos-steam://install) 
when i tried, it reports "Device not compatible"

I could install steam in a container but is there any way to install steam for chromeos like bypass google limitation?

My device is AMD Ryzen 7 8745H w/ Radeon 780M Graphics and 32GB ram with 1TB nvme (image size is 96GB).
i think that is enough...?

Githubguy132010, 2025-04-12:
No, minimum storage needs to be 128GB

SteveZMTstudios, 2025-04-12:
> No, minimum storage needs to be 128GB

I reinstalled the chromebook, build the image  that 192GB size and lost all my data, and now I'm sure it has nothing to do with size.
Because I still can't use insert coin or install steam.

closed 2025-11-10 by github-actions[bot]