sebanc/brunch #2223: Chrome OS Flex 135.0.7049.104 storage

xenobinary, 2025-04-27:
Today, I used the latest Brunch r135 stable 20250421 install image for Chrome OS Flex 135.0.7049.104 to create a 55GB image file on my host os (Ubuntu 24.04.2 LTS). All the installation processes went smoothly. However, I encountered two issues after installation:

1. I cannot access storage details via Settings -> System preferences -> Storage management.
2. I cannot install Linux, as Chrome OS reports only 2.3GB of storage remaining. Checking `chrome://system`, FREE_DISK-SIZE shows approximately 50GB available. However, the command `df -h` in the shell indicates only 2.3GB of available disk space for /dev/sdz1.

Returning to Ubuntu, I used` losetup -Pf chromeos.img` and confirmed the partition size was correct, with around 50GB remaining. Upon inspecting with GParted, it indicated over 50GB of unassigned space in stateful partition needing repair. Hesitant to use GParted directly, I used the command `resize2fs /dev/loopXpY`. 

After rebooting into ChromeOS Flex, the second issue was resolved; however, the fist issrue remains.

Thank you.

Anonyoumss, 2025-05-12:
Also you do not need a Linux you can go to tty2 or Crosh by either Ctrl + Alt + T and shell then: sudo resize-data or Ctrl + Alt +f2 login in as chronos then sudo resize-data

Anonyoumss, 2025-05-12:
Regarding that I also found a method to make the disk file bigger so I went to a Linux boot and used dd or truncate to enlarge my file then I made the partition bigger and I added 1 GB then it fixed itself and showed me my 20 GB free 

[image]
Issue 1 still not fixed

Anonyoumss, 2026-05-29:
I remember fixing this. It was kinda complex. I moved to a Linux distro and had to edit the actual .img file via a weird setup. I made an issue here, and ended up fixing it myself. My explanation was not detailed. But this is what I remember: I found free space on the img file that had .img files inside, so I used some sort of software to resize the data partition to fix it. Sebanc personally helped me solve it on Discord, but he could not figure out the whole thing. It was ideas until I figured it out. Now reinstalling fixes it, and it was in a specific r135 version.